free5lot/hid-apple-patched #1: Submit this to the kernel and get it in ASAP! :)

atrauzzi, 2015-02-26:
What are you waiting for? :)

free5lot, 2015-02-26:
Hi. Nice issue)
I was reading about rules and requirements about making a patch and pulling it, so it wouldn't be rejected by the maintainer.  It's a new thing for me and Jan (@JanmanX).
Are you familiar with sending patches to the kernel upstream?

This version of patch is quite simple and clean, so I do expect it to be admitted.

atrauzzi, 2015-02-26:
Unfortunately I'm not familiar with it, I'm just really hopeful it gets in.  Cannot believe it's happened sooner.

atrauzzi, 2015-02-26:
Out of curiosity, which one should I be using?  @janmanx's or yours?

free5lot, 2015-02-26:
Nice to hear it. I'm going to try within a week. The problem exists for years, so a week or two is quite small time for waiting)

BTW, have you tried the patched version?

free5lot, 2015-02-26:
Well, I would recommend this version, because it's a bit cleaner in code and is up to date (about kernel source).

free5lot, 2015-02-26:
Also, Jan (@janmanx) has also tested my patch and found it good and uses it now.
Anyway he was a pioneer of solving this issue!

If you install this patched version, please public your results, does it work for you and what device do you have.

atrauzzi, 2015-02-26:
Any plans to support DKMS?

free5lot, 2015-02-26:
Well, it's a small patch for hid-apple. None of us is not an author of hid-apple module, so the question about DKMS is probably irrelevant.
If this patch is admitted to the upstream kernel then there'll be not need for DKMS, will it be?

atrauzzi, 2015-02-26:
Yeah, I just figured it might be nice as it could be a while before the patch gets into any actually released kernels (Ubuntu).

free5lot, 2015-02-26:
I think, I'll update a Ubuntu's wiki here 
https://help.ubuntu.com/community/AppleKeyboard
with a simple build&install instructions from git repo.

JanmanX, 2015-02-28:
Good idea to update the Ubuntu's wiki. This might prove as a good testing ground for the patch, and even a proof of the quality of the patch :)   (The maintainers usually want tested code, or else they might reject it instantly)

I figure that if the patch indeed proves to be bug-free, we can try contacting some of the previous authors or some maintainers of the HID modules and see if they are kind enough to help us push it through to the kernel.

I am somewhat busy right now, but when I get the time, I am ready to try to make valid Linux kernel patch of the code and see if I can manage to get a hold of some Linux kernel hackers to help us out.

/Jan

free5lot, 2015-03-01:
Thanks everybody, I've added installation script, please take a look, it may have problems or simple typos as I've made it rather complicated and clever.
If it works OK for you, I'll update Ubuntu's wiki and send patch to maintainers.

free5lot, 2015-03-01:
I not only made installation script, but also made changes for passing kernel guidelines checker (scripts/checkpatch.pl in Linux kernel), and prepared a patch to fit this kernel checkpatch.pl requirements.
So, I'm going to send this patch soon to an appropriate maintainer.

JanmanX, 2015-03-02:
Great news!

Please keep us updated on the progress!

free5lot, 2015-05-01:
Sorry guys for being away for two months.
I'm still using this patch and everything seems to work fine, no more nightmare with different control placement on different devices for me :)

free5lot, 2015-05-01:
# This patch was submitted to the kernel maintainers today!
I hope it'll be accepted. Thanks to everybody how helped and especially to Jan.

JanmanX, 2015-05-01:
Amazing work!
Even with the slight hardware limitation on wireless devices (the hardware FN), I still think this patch will be widely used and enjoyed by many users!

atrauzzi, 2015-05-01:
Congrats guys!  If there's any advocacy/bumping you need done, feel free to link to the destinations here.

free5lot, 2015-05-02:
Thanks, guys! I made it like in patching rules - sent it by email (using real name as it's required by rules) to maintainers. I was put to mail-lists, I've found it here today:
https://lkml.org/lkml/2015/5/1/213
https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg878295.html
So, it'll be probably discussed there. I think, the main maintainer for hid-apple is Jiri Kosina (jkosina@suse.cz). Probably we will wait his response there.

danihodovic, 2016-01-01:
What is the status of the kernel patch?

free5lot, 2016-01-01:
1 of May 2015 -  the patch was sent to the maintainers according to all kernel code rules.
The responsible hid and hid-apple maintainer never replied there, neither did anyone else: https://lkml.org/lkml/2015/5/1/213

Since then the code was changed to fit newer kernels and new MacBooks, so in fact, not sure how it could get to the Linux upstream if patches are simply ignored there.
Any ideas?

danihodovic, 2016-01-02:
I have no clue on how the kernel contribution process works. I'm a little surprised that this doesn't get more attention though, are there that few people running Linux on Macbook Pro?

FYI: I tried this patch with 4.4.0-040400rc7-generic and it seems to be working fine. It seems like hid-apple.c didn't change between 4.2 and 4.4.0-040400rc7-generic.

p-kozlov, 2016-06-14:
@free5lot what is the status of current patch?
https://lkml.org/lkml/2015/5/1/213 - I can't see anything there?

free5lot, 2016-06-14:
> @free5lot what is the status of current patch?
> https://lkml.org/lkml/2015/5/1/213 - I can't see anything there?
Status is something like - "Ignored by maintainer".
About lkml.org link - I see nothing there, too, not sure why.

Patched version was sent to kernel's maillists, so it's mentioned online at many places, like: http://www.kernelhub.org/?msg=739170&p=2

Anyway, since then the patched moved forward and supports the new kernel with a few new features.
I'll be able to check it with the last LTS Ubuntu soon and maybe try another patch sending according to kernel's code-rules.

fancywriter, 2017-01-27:
For those who use Gentoo - feel free to use my patched ebuild for gentoo-sources here https://github.com/fancywriter/fancy-overlay/blob/master/sys-kernel/gentoo-sources/gentoo-sources-4.9.6.ebuild

Works not only as a separate module, but as built-in option to kernel (how I use it).

benjeffery, 2017-05-08:
Seems a shame this is being ignored - I was really pleased to find a solution and using DKMS it worked perfectly. Thanks!

fancywriter, 2017-07-02:
Should be in **gentoo-sources** from version 4.12 - one more reason to switch to Gentoo if someone is in doubt. https://gitweb.gentoo.org/proj/linux-patches.git/commit/?h=4.12&id=593358522f8e378eb4cc0e20efa2e57d6669adfd

vilian, 2017-07-15:
Works great with ubuntu 16, 4.4.0-83-generic on macbook late 2008 with french keyboard layout. This would've saved me some frustration 3 years ago!

free5lot, 2017-07-26:
Thanks @benjeffery , @fancywriter , @vilian and everyone else. I'll try to make an updated patch for mainstream Linux kernel and try to push it there again.

free5lot, 2017-12-12:
The patch is still working fine and in demand by people, so on holidays in January I'll try to make a version based on latest kernel code checking kernel code guidelines and will try to report it as a patch to mainstream one more time. I will keep you updated on this.

peerchemist, 2020-04-20:
Hi, are you still trying to get this into mainline?

free5lot, 2020-04-29:
@peerchemist today I sent letters to the responsible maintainers (according to get_maintainer.pl script):

> Jiri Kosina (maintainer:HID CORE LAYER)
> Benjamin Tissoires (maintainer:HID CORE LAYER)

and I hope at least  one of them will reply and I will be able to provide patches without being ignored. Currently I plan to provide patch for only swapping fn and ctrl. And if everything will go smooth then other patches that are included in this repo.

peerchemist, 2020-04-29:
That is the most important patch indeed. Okay, good luck.

free5lot, 2020-04-29:
Jiri Kosina replied. So the process started, maybe in a week it will be done.

free5lot, 2020-05-15:
The patch was submitted to the kernel's developers (Jiri in particular).
I will let you know more as the situation will be changing.

almson, 2021-08-20:
Did the patch get merged?

almson, 2021-08-20:
I see the patch was merged for Linux 5.8. Congrats! I've updated README in PR https://github.com/free5lot/hid-apple-patched/pull/75

Any plans to merge `rightalt_as_rightctrl`?

free5lot, 2021-08-20:
Yes, but I had no time or perseverance to prepare proper `rightalt_as_rightctrl` patch for the maintainer.
Are you willing to help with it, it will be great?

almson, 2021-08-23:
I've never done that. Can you tell me what's involved? Besides making and testing a minimal commit, of course.

captn3m0, 2023-01-18:
Working on creating a patch for `rightalt_as_rightctrl` against the AsahiLinux fork.

free5lot, 2023-01-19:
After the `swap_fn_leftctrl` feature was accepted upstream I was kind of satisfied enought with upstream hid-apple.
But maybe I should still send other patches from the project to upstream, it will make no need to implement `rightalt_as_rightctrl` in fork, even on M2, is it right? Or do they use completely different hid-apple sources?

captn3m0, 2023-01-19:
It's the same source, with a few extra commits to add M1 support. I did a diff locally, and it was mostly just magic numbers for the M1/M2 keyboards.

https://github.com/AsahiLinux/linux/commits/asahi/drivers/hid/hid-apple.c

free5lot, 2023-01-19:
Hmm, I see. So is it preferable to push patches to upstream ASAP and take them to the M1/M2 fork or you prefer to support this patches separately?

captn3m0, 2023-01-19:
It's preferable to push them upstream, as Asahi runs close to upstream, and can easily adopt new patches.

free5lot, 2023-09-21:
I tried to contact upstream maintainers that are responsible for `hid-apple` but nobody responded. The original maintainer that was available via email is not active for now. If somebody knows the better contact to any maintainer who can accept patches or PRs, please say. The ways changes are getting into the kernel do not seem optimal nowadays: submitted patches are ignored, maintainers are not available.

bam80, 2024-12-21:
Any update? Maybe try again?