free5lot/hid-apple-patched #2: Remapping not working

atrauzzi, 2015-02-26:
Just tried to run and set `swap_fn_leftctrl` to `1`.  Both control and fn stopped working.

atrauzzi, 2015-02-26:
Just a quick update, when I set it up for a permanent install however, it works.  I'm fn-ctrl swapped!

free5lot, 2015-02-27:
Hm, maybe something was wrong during your first attempt to run it.
Nice, it works anyway.
Please post your device information you use.

atrauzzi, 2015-02-27:
What's the format you'd prefer for that?

free5lot, 2015-02-27:
Something like device model, GNU/Linux version (distribution and kernel version).
+ Do you have Eject key?

atrauzzi, 2015-02-28:
Ubuntu 14.10
3.16.0-31-generic
MacBook Pro Late 2014 (MacBookPro11,2)

No eject key.  Power switch is in the top right of my keyboard, not a small round button.

free5lot, 2015-03-01:
Well. It means the patch was tested on two MacBooks Pro and works on both.

closed 2015-03-01 by free5lot