free5lot/hid-apple-patched #3: Add permanent install script from parent repo.

atrauzzi, 2015-02-27:
Minor item, just to make lives easier.

free5lot, 2015-02-27:
Accepted. Will be done rather soon.

free5lot, 2015-02-27:
And the install script is going to be a bit more complicated and enhanced, as the build-script is.

free5lot, 2015-03-01:
Installation script was added (install.h).
It is rather cleaver and complicated script, please take a look. It applies patched version with options on fly and if user sees everything works fine - it makes permanent system install.

JanmanX, 2015-03-01:
Your installation script works great, I like the "Does keyboard work as you expect? [Y/n].

Line 80-82, you try to save previous parameters, but you don't check if the module is loaded, in which case `/sys/module/hid_apple/` might not exist. Maybe place that code in your pre-check functions, where you check for existing apple module? (line 17-24)

Great job on the script!

/Jan

free5lot, 2015-03-01:
Well, I've though of this.
On the other hand VAL=`cat NOT_EXISTING_FILE` sets $VAL to nothing and puts message to error stream (on the screen). So user will see a message about it and maybe it's a better thing, than keep it in silence?
Also it only affects on-fly test (insmod) and doesn't matter at all after system is rebooted.

free5lot, 2015-03-01:
Oh, I have an idea, how to deal with it.
Thanks for pull request, I've merged it.

JanmanX, 2015-03-01:
Oh, just realized now that it only affects insmod, and not long term installation.
You're right that some feedback to the user is good, but maybe we should print our own message instead?
3 warnings from cat and 2 "ERROR" messages from can make a user panic, if he/she doesn't know if it is critical for the installation (like me).

free5lot, 2015-03-01:
Done, check it out.

closed 2015-03-02 by free5lot