free5lot/hid-apple-patched #6: Ctrl-Shift-T not working(!)

JanmanX, 2015-03-07:
When swapping Left-Shift with FN, CTRL-SHIFT-T  combination is not being registered. 
Using the original patch supplied in this repository (hid-apple.c.original_(kernel_3.16.0-28), the combination works.

I first noticed it when trying to reopen last opened tab in Chrome, but verified in some other programs. 
You might want to get some better readings using "xev".

Free5lot has a much better understanding of the module, but I figure it is a problem with the mechanical part of the keyboard, and not the patch itself, and thus, can't be solved. (@free5lot, please verify :) )

Also, the key-combination is not working either on my version of this patch.

free5lot, 2015-03-07:
NOT CONFIRMED.

Well, first of all I checked and it works for me. I've checked Ctrl+Shirt+T in Firefox and in Konsole (KDE). So, it's really and be a hardware problem of yours but I can't be sure.

Under "Ctrl" we both mean left-bottom key on keyboard with "fn" label on it, yes? :)

Secondly, I cloned the kernel from mainstream repo a week ago and checked, the hid-apple.c file is the same as this patch uses, so everything is up to date.

free5lot, 2015-03-07:
Jan, what do you exactly mean with "a problem with the mechanical part of the keyboard"? Is it broken or there is a hardware limitation of fn+2_more_keys events?

JanmanX, 2015-03-07:
@free5lot 
Yes, I press the "fn" which is mapped to ctrl. 
To be clear, I press buttons **labelled** with: FN-LSHIFT-T, which are **mapped** to LCTRL-LSHIFT-T.

By "problem with mechanical part of the keyboard" I meant a hardware limitation of the keyboard - yes, something like fn+2_more_key events.

I have verified this issue on 2 separate machines running Ubuntu 14.04, but with the **same** keyboard.

I will try to find any documentation on this FN key...

free5lot, 2015-03-07:
Well. Than good news is that Macbook Pro Retina doesn't have this limitation.
And bad one - maybe it'll be unsolvable for you if there is electric limitation on the keyboard's board.
One more chance - try LiveCD of Kubuntu or other KDE-based GNU/Linux. Maybe it's some kind of Gnome's limitation, but I doubt it.

What model of external keyboard to you have exactly?

JanmanX, 2015-03-07:
@free5lot 
I have the "Apple Wireless Keyboard" model: A1314 [Image](http://upload.wikimedia.org/wikipedia/commons/thumb/9/96/Apple-wireless-keyboard-aluminum-2007.jpg/320px-Apple-wireless-keyboard-aluminum-2007.jpg).

I have found no documentation on limitations of the FN key, but I have found a very strange behavior.
Using [this site](http://www.microsoft.com/appliedsciences/content/projects/KeyboardGhostingDemo.aspx), I can press any key with the combination of fn-shift (mapped to ctrl-shift) **except** "t".

Unfortunatelly, Ubuntu seems to be the only distro that supports my bluetooth drivers out of the box. Installing drivers for my adapter on Kubuntu is a mess, so I cant test it on KDE just yet.

free5lot, 2015-03-07:
@JanmanX 
> Using this site, I can press any key with the combination of fn-shift (mapped to ctrl-shift) except "t".
Quite interesting and strange! So Ctrl+Shift+… almost works for you except T char.
Please try ctrl+shift+… combinations somewhere in Gnomes soft. And maybe in settings where you can set hotkeys.

Maybe Fn+Shift+T means something exceptional on Mac.

JanmanX, 2015-03-07:
@free5lot 
Maybe. The Gnome Keyboard shortcut program doesn't register the combination either.

I cannot help any further anymore, as I am rather busy, but here is samle output from "xev" of the combination being pressed. Maybe you can make some sense of it:

```
KeyPress event, serial 37, synthetic NO, window 0x4c00001,
    root 0x2c9, subw 0x0, time 3498322, (16,7), root:(16,59),
    state 0x0, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x4c00001,
    root 0x2c9, subw 0x0, time 3498772, (16,7), root:(16,59),
    state 0x1, keycode 37 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

-------------------------------- "T" is being pressed, but xev reports KeyRelease event on Left-Shift
KeyRelease event, serial 37, synthetic NO, window 0x4c00001,
    root 0x2c9, subw 0x0, time 3500772, (16,7), root:(16,59),
    state 0x5, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

-------------------------------- "T" being released, xev reports KeyPress on Left Shift
KeyPress event, serial 37, synthetic NO, window 0x4c00001,
    root 0x2c9, subw 0x0, time 3501360, (16,7), root:(16,59),
    state 0x4, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

```

free5lot, 2015-05-01:
So, it seems to be some hardware related problem of external keyboard, and doesn't affect MacBook Pro or other devices, that can get key-press correctly (how doesn't make FN key (Ctrl after modification) as something special on HARDWARE level). So it's not a patch issue and probably can't be fixed for this hardware.
So I'm closing this ticket as on other hardware it was not confirmed.

closed 2015-05-01 by free5lot

migmartri, 2016-08-08:
I can confirm the issue with the "Apple Wireless Keyboard" model: A1314.

@JanmanX did you figure something out?

muesli, 2017-02-22:
Same behavior with Wireless Keyboard A1255.

free5lot, 2017-02-23:
I have an question to guys, who have original MacOS and this external Apple Wireless Keyboards. Can you switch the buttons in MacOS and try ctrl+shift+t there? It may either prove it's a physical limitation or give some hope.

StephanDollberg, 2017-09-19:
Can also confirm with A1644. Only T doesn't work. Everything else does.

free5lot, 2017-09-22:
@StephanDollberg  Can you switch the Fn and LeftCtrl  buttons in MacOS and try Ctrl+Shift+T there (really Fn+Shift+T actual keys)? Will it work there?

StephanDollberg, 2017-09-22:
@free5lot I don't have access to a Mac right now but I might have in a couple weeks (not fully sure yet though). Then I can try.

free5lot, 2019-02-06:
Some additional information from @djdevin:

> FWIW if you train yourself you might be able to get around this limitation. The keys cannot simultaneously be pressed but you have a small window which may be recognize as simultaneous.
> 
> "Roll" up to Control-Shift-T by pressing no more than 2 keys at the same time (control, then shift, then start to pull off as you press T) and you should be able to trick it into thinking you pressed them all together. I accidentally discovered this and I can get it 99% of the time now.
> 
> I don't have monster hands so I can't do it with 1 hand which might be more difficult.

klaeufer, 2019-04-10:
Just for the record:

I am using this configuration on an external slim Apple Bluetooth keyboard (A1314):

```
options hid_apple fnmode=2
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1
```

- Control_L Shift_L t works only with "rolling" off the Control_L key as I am pressing the t key
- Control_L Shift_R t works perfectly
- Control_R Shift_R t works perfectly (I can live with this one for reopening closed tabs)
- Control_R Shift_L t works perfectly

Other three-key combinations also work perfectly, e.g.:

- Control_L Shift_L TAB
- Control_L Shift_L c
- Control_L Shift_L v

In sum, I'm grateful for this clean solution for matching my external Apple keyboard to the one built into my Dell XPS laptop.

free5lot, 2020-03-31:
@almson can you please test this combination using Bluetooth connection mode?

almson, 2020-03-31:
Ctrl+Shift+T doesn't work in Bluetooth mode either, but the trick with "rolling" works! Thanks!

free5lot, 2020-03-31:
So, it's still a hardware limitation.
Thanks for information.

almson, 2020-03-31:
The entire 5-T-G-B column is affected. Fn+Option+G doesn't register (to use the silkscreened names), as well as Fn+Command+B, and Fn+Control+5.

free5lot, 2020-04-01:
>The entire 5-T-G-B column is affected.
Oh, that makes more sense then. That indeed seems to be cutting corners by Apple.
Maybe these details should be mentioned at the end of README?

PS. On Macbook Pro there is no issue with Ctrl+Shift+T and probably any issue with any column at all.

Hubro, 2020-06-01:
This also affects me using the Magic Keyboard A1644.

[image]

A quick tip to those wondering how to do the "rolling trick". Timing doesn't matter. Just press "Fn+Shift+T", then release the Fn key first. That will always register a Ctrl+Shift+T press.

free5lot, 2020-06-08:
>A quick tip to those wondering how to do the "rolling trick". Timing doesn't matter. Just press "Fn+Shift+T", then release the Fn key first. That will always register a Ctrl+Shift+T press.

Thanks for the tip. I will add it to the Readme.

vasilakisfil, 2020-06-16:
I can confirm that this combination doesn't work in A1644 (nor usb-connected, neither bluetooth connected) and it didn't work in A1242 either, which I had been using for ages. The rolling trick is super helpful though, thanks for the tip! :+1: for adding it to the Readme.

jc00ke, 2020-09-29:
Wow, I don't know how @klaeufer found the "rolling-off" trick but thank you! :pray:

free5lot, 2020-09-29:
@jc00ke well, actually @djdevin was the one who provided information about rolling trick. Plus @almson and @Hubro made the situation more clear and provided the way of using it with 100% success.

jc00ke, 2020-09-29:
My bad! Thanks all around!

willmo, 2023-03-20:
FWIW, not sure if it's already known, but in testing with A1644 (ANSI US English) connected via USB to macOS 13.2.1, and mapping the physical fn key to ctrl, I can reproduce the issue with T, but not with 5-G-B. So it's at least partly an OS-independent issue...