free5lot/hid-apple-patched #9: Doesn't not work with MacBook Pro (Retina, 13-inch, Early 2015)

Aetf, 2015-09-24:
The keycode for Fn key is 464(0x1d0) on my macbook pro, not 84.

After changing [Line 40](https://github.com/free5lot/hid-apple-patched/blob/master/hid-apple.c#L40), the module works as expected.

The model identifier is MacbookPro12,1 if that is needed.

free5lot, 2015-09-28:
Hm, thank you @Aetf for this information, it's quite impotent.
Please post your GNU/Linux distribution, version and kernel version here.

Aetf, 2015-09-30:
Distribution: Archlinux
Kernel: mainline kernel with this patch applied
Output of `uname -a`
```
Linux Aetf-Arch-Mac 4.3.0-rc2-mainline #1 SMP PREEMPT Wed Sep 23 14:35:46 EDT 2015 x86_64 GNU/Linux
```

And for the key code, I'm following the instruction on [ArchWiki](https://wiki.archlinux.org/index.php/Extra_keyboard_keys#Using_showkey)

free5lot, 2015-10-02:
Maybe the difference in codes between patched version and not-patched is because of something being changed in kernel with file like hid-ids.h.
I've been pushing this patch to mainstream (links in issue #1) and it was built on the latest kernel code with all code requirements for kernel patches, but it was ignored by hid-apple maintainer and may become somehow outdated.
I don't know what to do in this situation as for now I don't have enough time to figure it out and don't have 2015's Macbook Pro (if it's device-related thing). If someone has any suggestions please give me a message.

Aetf, 2015-10-04:
More investigation shows that the driver support for 2015's Macbook Pro was added some version between 4.0.1 and 4.2, so Fn key on 2015's MBP is recognized with a standard key code KEY_FN(464). So no need for a special key code 84. However, this should also be the case for earlier models, not sure why :/

For the problem that when swap_fn_leftctrl disabled, Fn key not working properly, I think I have found the reason.

There should be only one place dealing with changed fn keycode and other parts should be left unmodified. The function [`apple_input_mapping`](https://github.com/free5lot/hid-apple-patched/blob/master/hid-apple.c#L384) is called before `hidinput_apple_event` each time a key pressed. Thus modifying function `apple_input_mapping` isn't necessary.

If revert `apple_input_mapping` to original version, the patch works perfectly both when swap_fn_leftctrl enabled or disabled. Tested using Archlinux stable kernel version 4.2.2.

Aetf, 2015-10-04:
I have created a PR #10 with the bug fixed and updated against kernel version 4.2.2

free5lot, 2015-10-04:
Thank you, @Aetf, great job!
If anything works as expected after fix then I'm accepting you PR, thanks.

closed 2015-10-04 by free5lot