free5lot/hid-apple-patched #18: DKMS support

almson, 2016-03-12:
DKMS support would be very handy.

Here is a `dkms.conf` file that seems to work:

    MAKE="make LINUX_HEADER_DIR=/usr/src/linux-headers-${kernelver}"
    CLEAN="make clean"
    BUILT_MODULE_NAME[0]=hid-apple
    DEST_MODULE_LOCATION[0]='/kernel/drivers/hid'
    PACKAGE_NAME=hid-apple
    PACKAGE_VERSION=1.0
    REMAKE_INITRD=yes

To install, save the `dkms.conf` file into the source code folder, and:

    sudo dkms add .
    sudo dkms build hid-apple/1.0
    sudo dkms install hid-apple/1.0

The instructions to `override hid_apple * extra` are not needed.

free5lot, 2016-03-17:
Thank you @almson for your issue, could you make this as Pull Request so I could accept your tested version?
PS. Maybe AUTOINSTALL=yes is a good option too, and maybe MAKE and CLEAN options are not needed, are they?

almson, 2016-03-21:
Ok, I will do that after I test it thoroughly. I haven't worked with dkms before, so I'm not familiar with all the options. I'll test your suggestions.

free5lot, 2016-03-24:
Thank you @almson.

almson, 2016-05-16:
I've updated the dkms config in the original post and confirm that it's working great through many kernel updates.

free5lot, 2016-05-16:
Thank you @almson, I'll add this as new file and will add it to instruction.
Do you personally consider it as the best install option over install.sh and manual build in Ubuntu?

almson, 2016-05-17:
I think dkms is easy and you only have to do it once. Again, I'm not too knowledgeable about these things to know what advantages the other approaches might have. Perhaps the linux header path has to be adjusted on other distros?

What would be the *best* install option on Ubuntu, however, would be a .deb file that packages the dkms, is in the repos, and shows up in the `Additional Drivers` applet. Then we could tell Linus and his kernel team to kiss our ass.

Some instructions: https://help.ubuntu.com/community/Kernel/DkmsDriverPackage If I have time I'll give it a shot to try to create it.

free5lot, 2016-05-18:
It would be great. But not sure if it's not an overkill for a patch with several additional kernel options.

Ichigo-Roku, 2016-05-28:
After doing the "dkms install", what do I need to do to swap fn and control? I rebooted my computer but nothing changed, I think that I need to change some parameters in a file or something, but I can't find it.

Thanks for your help!

free5lot, 2016-06-01:
Hm, this information should be added to README for DKMS-installation way.

Not sure about any difference for DKMS installation, but in usual case you should modify modprobe config file of hid-apple, like /etc/modprobe.d/hid_apple.conf and add desired options there.
E.g. options like swap_fn_leftctrl=1, ejectcd_as_delete=1. 
Usual hid-apple options fnmode, iso_layout and swap_opt_cmd are also supported.

Additional available options of this patched version:
- swap_fn_leftctrl - Swap the Fn and left Control keys. (For people who want to keep PC keyboard muscle memory. [0] = as-is, Mac layout, 1 = swapped, PC layout).
- rightalt_as_rightctrl - Use the right Alt key as a right Ctrl key ([0] = as-is, Mac layout. 1 = Right Alt is right Ctrl)
- ejectcd_as_delete - Use Eject-CD key as Delete key. ([0] = disabled, 1 = enabled)

More information how to add these options:
https://help.ubuntu.com/community/AppleKeyboard#Ubuntu_11.10_up_to_recent
https://wiki.archlinux.org/index.php/Apple_Keyboard

Ichigo-Roku, 2016-06-01:
After looking at the install.sh file I've found out the file to modify, but it still didn't work, I missed that I needed to do a: "sudo update-initramfs -u -k all" like mentioned in your first link. It was my mistake.

Thanks for your help!

free5lot, 2016-06-14:
@Ichigo-Roku you're welcome!

closed 2016-08-03 by free5lot