free5lot/hid-apple-patched #20: Add mappings for Pause, Break, ScrLK, SysRq

almson, 2016-03-21:
Recent Lenovo keyboards copy Apple by omitting Pause, Break, ScrLK, and SysRq keys. Since these keys are sometimes necessary for PCs (unlike for Macs), they're emulated using Fn+ combinations. I propose to support these same combinations in the driver without a configurable option, since I don't believe there exists a competing way to map them.

    Break (Fn + B)
    SysRq (Fn + S)
    ScrLK (Fn + K)
    Pause (Fn + P)

Reference: https://support.lenovo.com/ua/uk/documents/ht074004

almson, 2016-03-21:
Actually, Dell keyboards also sometimes don't have these keys. Dell didn't settle on a standard mapping. Looks like they're all over the place with different models. I would still adopt the Lenovo standard.

Aetf, 2016-03-22:
I would say we are making the driver a fully programmable one by adding more mapping options like this :smile:. Definitely we should get the underlying remapping logic right before this.

free5lot, 2017-06-11:
Well, hid-apple should be about apple keyboards, right? It seems not the best idea to maintain the lenovo keyboards in this module anyway :)

almson, 2021-08-20:
The real reason for these keys is to use Magic SysRq.

Due to stupid Linux OOM handling, I deal with OOM freezes quite often, but Magic SysRq also has many other uses..

fvsnippets, 2023-10-08:
Hi!
For the Magic SysRq issue, I am using a workaround that mixes `udev`, `xmodmap` (I know that this won't work on Wayland but there are alternatives for that... but I don't use wayland, yet) and `loadkeys`.

Basically what I do is:
1) I remap my *CapsLock* physical key to *sysrq* keycode using `udev`. As far as I understand the kernel will know about this "mapping", so It can known when both "SysRq" (CapsLock) and "Alt" key are pressed in conjunction. It wil "take" the event and wont propagate it to the rest of the system (i.e. you can't use Alt+SysRq [CapsLock] for other actions).
2) I (re)remap *sysrq* `xmodmap` keycode to *Caps_Lock* keysym. This is a mapping only know by the X environment.
3) I (re)remap *sysrq* `loadkeys` keycode to *Caps_Lock* keysym. This is a mapping only know by the "consoles" (TTYs) environment.

So, CapsLock will continue acting as a normal CapsLock key without Alt being pressed. But with Alt pressed, CapsLock will act as SysRq.
As far as you don't need to use the Alt-CapsLock combination for other things,  then this workaround should be enough.