free5lot/hid-apple-patched #23: Ubuntu 16.04, make install leads to ssl error

hickeroar, 2016-05-09:
```
root@geekbook:/home/ryan/Development/hid-apple-patched# export LINUX_HEADER_DIR=/usr/src/linux-headers-4.4.0-22-generic/

root@geekbook:/home/ryan/Development/hid-apple-patched# make
make -C /usr/src/linux-headers-4.4.0-22-generic/ M=/home/ryan/Development/hid-apple-patched modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-22-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-22-generic'

root@geekbook:/home/ryan/Development/hid-apple-patched# make install
make -C /usr/src/linux-headers-4.4.0-22-generic/ M=/home/ryan/Development/hid-apple-patched modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-22-generic'
  INSTALL /home/ryan/Development/hid-apple-patched/hid-apple.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-22-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-22-generic'

root@geekbook:/home/ryan/Development/hid-apple-patched# 
```

Anything obviously wrong with what I'm doing here?

JanmanX, 2016-05-10:
The new Ubuntu version uses the Linux signed kernel modules. When you compile a module, it will thus sign it with your keypair.

Maybe this can be turned off, but maybe you can just create a keypair and try again?
`ssh-keygen -t rsa`

free5lot, 2016-05-11:
I'm new to this signed thing, so if someone practically resolve this issue, please make a post or PR for it. Thank you.

hickeroar, 2016-05-11:
I already have ssh keys generated. This seems to be looking for a signing certificate, and not a ssh key pair. I've attempted to generate a certificate, but I seem to be doing something wrong. The certificate is generated and "should" be working, but it doesn't seem to have any effect. I'm probably screwing something up.

almson, 2016-05-16:
I installed on Ubuntu 16.04 using dkms (see here: https://github.com/free5lot/hid-apple-patched/issues/18) without problems.

free5lot, 2016-05-16:
@hickeroar did dkms approach work for you?

almson, 2016-05-24:
I found out a couple things:

1. MacBooks don't support secureboot (AFAIK), and module signing is a non-issue except, I guess, for PC users of the apple keyboard. (This is true on my MBP and on Ubuntu's latest kernel. It's possible another kernel might demand module signing even without secure boot enabled.)
2. Signing your own modules is a pain. See: http://askubuntu.com/a/773852/463546 and: http://us.download.nvidia.com/XFree86/Linux-x86/361.42/README/installdriver.html#modulesigning

There are 3 quick solutions:

1. The secure method: Generate temporary keys to sign the module, import the public key with `mokutil` (which requires confirming the action during a reboot to prove that the user has physical control over the machine), and delete the private key when you're done.
2. The (slightly) insecure method 1: Disable secure boot
3. The (slightly) less insecure method 2: Disable module signature verification with `sudo mokutil --disable-validation` and a reboot (ignore the "Failed to request" error).

And 2 longer-term solutions:

1. Create a DKMS script that automates solution 1.
2. Somehow get this module into Ubuntu repos and have Canonical sign it.
3. Find Linus in an alleyway and beat him until he concedes that the process for including patches in the kernel is deeply flawed and doesn't scale.

free5lot, 2016-06-01:
Thank you @almson, useful overview of solution. Considering the alleyway, who has any experience of pushing patches to upstream in cases when the accountable maintainer doesn't response for patch offered?

samiralajmovic, 2016-06-21:
Like to add that installation via .sh worked for me and the dkms solution didn't. I'm running Ubuntu 16.04.

$ uname -a

Linux mac-name 4.4.0-24-generic #43-Ubuntu SMP Wed Jun 8 19:27:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

free5lot, 2016-06-22:
Interesting, so both ways work and don't work in some cases.

samiralajmovic, 2016-06-22:
I did attempt the dkms solution at first so might been a mix or something? When it was running in the console, I saw the .sh solution added 2 rows to hid-apple.conf file, does dkms work the same?

free5lot, 2016-06-22:
@samiralajmovic no, dkms does not, you should add it manually according to your desires. That's probably why you didn't succeed with dkms-way installation.

adityaputatunda, 2017-03-13:
Hello @JanmanX 
I tried to save a new key. But I don't know how to proceed with my issue. Please look here. [https://github.com/patjak/bcwc_pcie/issues/70](url) comment. Can you please help me with the issue?

free5lot, 2017-06-11:
DKMS installation way is considered as recommended now in Ubuntu, this this issue is not relevant anymore. I'm closing it, but free to add anything if you want.

closed 2017-06-11 by free5lot

chespinoza, 2018-04-25:
Maybe would be good to refer that DKMS is the prefered way for Ubuntu in the documentation.

free5lot, 2018-05-13:
@chespinoza, installation via DKMS is already marked as recommended way in `README.md`, what documentation do you mean?

p-himik, 2019-03-20:
@free5lot So is the signing supposed to be not required if I install the module via DKMS? Because I did just that (although I've changed some code) and none of the swaps work and `dmesg` gives me `PKCS#7 signature not signed with a trusted key`.

p-himik, 2019-03-20:
Please excuse the noise - Secure Boot is disabled on my system, and yet I see that error. Must be something wrong with my particular setup, but I have no idea what.

poseidon-j, 2020-06-14:
ssl error

Check old driver and unload it.
rmmod r8168
Build the module and install
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
Warning: modules_install: missing 'System.map' file. Skipping depmod.
DEPMOD 5.4.0-37-generic
load module r8168
Updating initramfs. Please wait.
update-initramfs: Generating /boot/initrd.img-5.4.0-37-generic
Completed.