free5lot/hid-apple-patched #24: How to apply this patch to the kernel source code

p-kozlov, 2016-06-14:
I am usually build kernel from the source code (gentoo-sources package) - and compile hid_apple inside kernel (not as a module).
Is it possible to apply some magic command to the source directory /usr/src/linux to modify vanilla hid_apple module to support swapping FN and CTRL keys? Would be great!

Or what I need to do to use it as a module - isn't it easier to just turn off hid_apply on menuconfig while building kernel and than compile this module?

p-kozlov, 2016-06-14:
Here is diff command output
```
diff hid-apple.c /usr/src/linux/drivers/hid/hid-apple.c
55,70d54
< static unsigned int swap_fn_leftctrl;
< module_param(swap_fn_leftctrl, uint, 0644);
< MODULE_PARM_DESC(swap_fn_leftctrl, "Swap the Fn and left Control keys. "
<               "(For people who want to keep PC keyboard muscle memory. "
<               "[0] = as-is, Mac layout, 1 = swapped, PC layout)");
< 
< static unsigned int rightalt_as_rightctrl;
< module_param(rightalt_as_rightctrl, uint, 0644);
< MODULE_PARM_DESC(rightalt_as_rightctrl, "Use the right Alt key as a right Ctrl key. "
<               "[0] = as-is, Mac layout. 1 = Right Alt is right Ctrl");
< 
< static unsigned int ejectcd_as_delete;
< module_param(ejectcd_as_delete, uint, 0644);
< MODULE_PARM_DESC(ejectcd_as_delete, "Use Eject-CD key as Delete key. "
<               "([0] = disabled, 1 = enabled)");
< 
183,197d166
< static const struct apple_key_translation swapped_fn_leftctrl_keys[] = {
<       { KEY_FN, KEY_LEFTCTRL },
<       { }
< };
< 
< static const struct apple_key_translation rightalt_as_rightctrl_keys[] = {
<       { KEY_RIGHTALT, KEY_RIGHTCTRL },
<       { }
< };
< 
< static const struct apple_key_translation ejectcd_as_delete_keys[] = {
<       { KEY_EJECTCD,  KEY_DELETE },
<       { }
< };
< 
217,219c186
<       u16 fn_keycode = (swap_fn_leftctrl) ? (KEY_LEFTCTRL) : (KEY_FN);
< 
<       if (usage->code == fn_keycode) {
---
>       if (usage->code == KEY_FN) {
221c188
<               input_event(input, usage->type, KEY_FN, value);
---
>               input_event(input, usage->type, usage->code, value);
300,323d266
<       if (swap_fn_leftctrl) {
<               trans = apple_find_translation(swapped_fn_leftctrl_keys, usage->code);
<               if (trans) {
<                       input_event(input, usage->type, trans->to, value);
<                       return 1;
<               }
<       }
< 
<       if (ejectcd_as_delete) {
<               trans = apple_find_translation(ejectcd_as_delete_keys, usage->code);
<               if (trans) {
<                       input_event(input, usage->type, trans->to, value);
<                       return 1;
<               }
<       }
< 
<       if (rightalt_as_rightctrl) {
<               trans = apple_find_translation(rightalt_as_rightctrl_keys, usage->code);
<               if (trans) {
<                       input_event(input, usage->type, trans->to, value);
<                       return 1;
<               }
<       }
< 
387,401d329
< 
<       if (swap_fn_leftctrl) {
<               for (trans = swapped_fn_leftctrl_keys; trans->from; trans++)
<                       set_bit(trans->to, input->keybit);
<       }
< 
<       if (ejectcd_as_delete) {
<               for (trans = ejectcd_as_delete_keys; trans->from; trans++)
<                       set_bit(trans->to, input->keybit);
<       }
< 
<         if (rightalt_as_rightctrl) {
<               for (trans = rightalt_as_rightctrl_keys; trans->from; trans++)
<                       set_bit(trans->to, input->keybit);
<       }

```
I already use swap_opt_cmd=1 and fnmode=2 options - can it break them or will everything works fine? I need all of the swappings: fn + left ctrl, cmd + alt, fn + functional keys.

free5lot, 2016-06-14:
Hi, Pavel!

First of all, **swap_opt_cmd** and **fnmode** options are also supported in the patched version with the same names, so there should be no problem.

About putting patched version to the kernel and building with it and not separately - well, it seems to be possible, why not, but no one will give any promises about it, neither somebody checked it. I can't help, not confident in your case.

p-kozlov, 2016-06-14:
@free5lot thank you - I tried today. Everything works fine - I just replace hid-apple.c in my kernel source directory and built it - works perfectly with building it inside kernel (not module).

I will think about better solution (like patch file) and update README.md with my information.

closed 2016-08-03 by free5lot

fancywriter, 2017-01-27:
I have created patch based on this source code and added it here https://github.com/fancywriter/fancy-overlay/blob/master/sys-kernel/gentoo-sources/files/gentoo-sources-hid-apple.patch - could be useful to anyone.