free5lot/hid-apple-patched #28: Add option to allow fn to function as additional control key, and not modify existing control key

aarobc, 2016-09-26:


free5lot, 2016-10-09:
Hi @aarobc. It may deserve consideration, but how do you use Fn key then? How to press Delete (Fn+Backspace), Home, Page Up?

aarobc, 2016-10-09:
Indeed I'd lose that functionality. Ideally I'd like to be able to have the remap apply only if there is no external keyboard attached. Perhaps set up a udev rule. 

Are there commands to enable/disable the fn remap without having to reboot?

free5lot, 2016-10-10:
> Are there commands to enable/disable the fn remap without having to reboot?

Yes, look in install.sh for a sample of use. Something like:
Reload with options
```
sudo rmmod "$MODULE_LSMOD_NAME"
sudo insmod "./$MODULE_FILENAME" $SAVED_OPTIONS swap_fn_leftctrl="$SWAP_FN_LEFTCTRL" ejectcd_as_delete="$EJECTCD_AS_DELETE"
Or change options without reload:
echo "$SWAP_FN_LEFTCTRL"  | sudo tee "/sys/module/hid_apple/parameters/swap_fn_leftctrl"
echo "$EJECTCD_AS_DELETE" | sudo tee "/sys/module/hid_apple/parameters/ejectcd_as_delete"
```

Where $XXX - are variables, you can replace it with fixed strings.

aarobc, 2016-10-11:
This is a suitable solution. Thanks for your assistance.

closed 2016-10-11 by aarobc

free5lot, 2016-10-12:
You're welcome. Happy, it helped.