free5lot/hid-apple-patched #32: iso_layout does not have an effect

Hannoskaj, 2017-04-03:
iso_layout has no effect in my configuration, French keyboard.

Hence the </> and @/# are swapped (or ²/³ if I configure the map to usual French PC keyboard).

All the other functions seem to work well (thanks for fn/ctl !).

Hardware:
MacBookPro11.4
French keyboard

Software:
Linux Mint 18.1 Mate with Cinammon as DE. (base Ubuntu 16.04)
hid-apple-patched commit feb734c4559c7cc8bfef9d8e8c3893eafe640ef7 (current)

I have installed the patch via dkms, following the instructions on github.
I have created the following configuration file as /etc/modprobe.d/hid_apple.conf
`
options hid_apple fnmode=1

options hid_apple swap_fn_leftctrl=1

options hid_apple swap_opt_cmd=1

options hid_apple rightalt_as_rightctrl=0

options hid_apple ejectcd_as_delete=1

options hid_apple iso_layout=0
`

Everything works, save the layout. No effect either when switching iso_layout to 1 and updating initramfs and rebooting. No effect either from setting 0 or 1 by the commands:
`echo 1 | sudo tee /sys/module/hid_apple/parameters/iso_layout`

The problem might be upstream, on the usual hid_apple?

Jonas

free5lot, 2017-04-06:
Hi @Hannoskaj, I'm happy that swap patch worked for you, you're welcome!

About iso_layout, can you please try the original hid-apple module on the same configuration with your French keyboard? Does iso_layout change anything there?

free5lot, 2017-04-06:
With a further analysis I have two things to say:
1) It seems iso_layout code is rather separated and should not be affected by patch. 
2) iso_layout changes anything only if (asc->quirks & APPLE_ISO_KEYBOARD) is true, so, if you check your case on the original hid-apple and it works the same (ignores the iso_layout flag), then it shouldn't be fixed in this patch.

PS. By the way, iso_layout only swaps 2 keys: KEY_GRAVE and KEY_102ND. You should check these 2 keys.

Hannoskaj, 2017-04-06:
Thank you @free5lot.
I would do the test, but I don't know how to revert to the vanilla module short of reinstalling…

That said, I  was surprised with the KEY_102ND, so I have looked into the xkb/keycodes files. Apparently the keycodes for </> and @/# are 94 and 49 (namely <TLDE> and <LSGT>). Hence they should not be the ones affected by iso_layout.

When looking back at Arch's wiki for Apple keyboard, they say that the apple:badmap xkb option corrects those two keycodes. Indeed it does. 

So at least for me, the question is solved. I probably should edit Arch's wiki, now!
Thanks again.

free5lot, 2017-04-09:
You're welcome.
I'll mark the issue solved then. You should feel free to post here the results of test for original hid_apple on your French layout if/when you manage to do it or to reopen the issue if your problem is still need assistance.

closed 2017-04-09 by free5lot