free5lot/hid-apple-patched #35: Toggleable Fn key

MageJohn, 2017-06-17:
I don't exactly know if this is in scope for this project, but is it possible to add a fourth fnmode which makes the Fn key toggle? Most of the time I'd like to use the media keys without the Fn key, but when I'm using an application with shortcuts mapped to the function keys I'd like to not have to hold the Fn key every time, 

Changing the fnmode manually is actually a bit of a hassle, requiring sudo and at least two or three commands.

free5lot, 2017-06-23:
Interesting idea, is it required by someone else? The discussion is welcome.

acheronfail, 2019-12-21:
Most Dells these days have the <kbd>fn</kbd>+<kbd>esc</kbd> key combination which swaps the function keys, I'd like this behaviour too.

For example, pressing <kbd>fn</kbd>+<kbd>esc</kbd> would swap the `fnmode` setting between `1` and `2`.

free5lot, 2019-12-22:
@acheronfail Thanks for this information and PR. I will take a deeper look and review it during holidays.

free5lot, 2020-01-12:
I'm in process of playing with current mainline kernel (5.5.0-rc5). If everything goes fine, I will try to push the patches upstream once more. Starting with the main one (about fn-ctrl swap) all others if it goes well.