free5lot/hid-apple-patched #36: Not working on Ubuntu 17.04 (4.10.0-24-generic)

tbjornli, 2017-06-23:
Is it only me or doesn't this work for Ubuntu 17.04 (4.10.0-24-generic) ?
I've searched through the issues, but couldn't find any related to this.

I've installed it via DKMS, but status give me the follow message:
```
$ dkms status
hid-apple, 1.0, 4.10.0-24-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```

I've tried to uninstall and remove the module, clone the repository and reinstall the module, but it still doesn't work and `dkms status` gives the same warnings.

It seems like several options are missing under */sys/modules/hid_apple/*
```
$ ls /sys/module/hid_apple/parameters/
fnmode  iso_layout  swap_opt_cmd
```

Any ideas?

free5lot, 2017-06-28:
No idea why 17.04 should work different than 16.04 LTS which works fine. Any 17.04 users are welcome to report their experience.
@tbjornli I can offer you to try dkms.conf from the previous commits, maybe it'll make any difference for you or search for similar problem in the internet.

tbjornli, 2017-06-29:
I have managed to solve the problem now. 

I uninstalled the module in order to try to reinstall with the previous dkms.conf and I noticed an error message while installing which I must have missed before.
```
$ sudo dkms install hid-apple/1.0 

hid-apple:
Running module version sanity check.
Error! Module version E18CE312FDCB68034845C15 for hid-apple.ko
is not newer than what is already found in kernel 4.10.0-26-generic (F8F097919D0FCE0DF2EE660).
You may override by specifying --force.

depmod...

Backing up initrd.img-4.10.0-26-generic to /boot/initrd.img-4.10.0-26-generic.old-dkms
Making new initrd.img-4.10.0-26-generic
(If next boot fails, revert to initrd.img-4.10.0-26-generic.old-dkms image)
update-initramfs......

DKMS: install completed.
```
In order to solve this problem, I reinstalled the module with the `--force` flag. 
```
$ sudo dkms install hid-apple/1.0 --force

hid-apple:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-26-generic/updates/dkms/

depmod...

Backing up initrd.img-4.10.0-26-generic to /boot/initrd.img-4.10.0-26-generic.old-dkms
Making new initrd.img-4.10.0-26-generic
(If next boot fails, revert to initrd.img-4.10.0-26-generic.old-dkms image)
update-initramfs......

DKMS: install completed.
```
And after a quick reboot, everything was working as normal again. I've since then reinstalled the module again using the latest dkms.conf and it works like a charm.

P.S. I found the solution after upgrading to kernel 4.10.0-26-generic, but I see now reason why it should not work for 4.10.0-24-generic as well.

closed 2017-06-29 by tbjornli

free5lot, 2017-07-01:
Thank you @tbjornli for detailed description. I will take this into account and maybe will be able to use this.