free5lot/hid-apple-patched #37: rmmod: ERROR: Module hid_apple is not currently loaded

ksantr, 2017-07-24:
I suppose keyboard uses a hid module, not hid_apple which is missing actually.
~$ lsmod | grep apple
```
hid_apple              16384  0
hid                   118784  4 hid_generic,usbhid,hidp,hid_apple
```

~$ modinfo hid
```
filename:       /lib/modules/4.10.0-27-generic/kernel/drivers/hid/hid.ko
license:        GPL
author:         Jiri Kosina
author:         Vojtech Pavlik
author:         Andreas Gal
srcversion:     E63C31BEC30B5CFB034E4B4
depends:        
intree:         Y
vermagic:       4.10.0-27-generic SMP mod_unload 
parm:           debug:toggle HID debugging messages (int)
parm:           ignore_special_drivers:Ignore any special drivers and handle all devices by generic driver (int)

```

free5lot, 2017-07-26:
What apple notebook or keyboard do you use?

ksantr, 2017-07-27:
Sorry, ubuntu 16.04, magic keyboard 2

free5lot, 2017-07-28:
Brief search gave me nothing. There are some external apple keyboards users here, maybe they know better about such situation. 
To be clear, in your case the apple keyboard of yours was managed by hid module (not hid-apple) before installation anything from this repo, wasn't it?

ksantr, 2017-07-29:
Yes, I'm sure. I had no even the hid_apple folder before installation.

free5lot, 2017-07-29:
@ksantr OK, then the patch didn't break anything. If you find how to make your external keyboard work with hid-apple then you're welcome to post it here, I'll add it to readme. 
I'm not a big help with this issue, it's not directly connected to the patched version of hid-apple. Try to search on the internet about it

ksantr, 2017-07-31:
OK, I'll try. Thanks.

closed 2017-08-04 by free5lot