free5lot/hid-apple-patched #42: Updates fopr 4.14.4?

Ndolam, 2017-12-10:
The hid-apple.c file in the kernel source has changed somewhere between 4.11.9 and 4.14.4.
On the other hand, the version here has not been changed.
Can anyone comment on either
(a) have you used the one here with 4.14.? with no apparent problems, or
(b) has anyone merged the kernel version updates into this?

Thanks.

free5lot, 2017-12-11:
It should probably work, but I should take a look and analyze the mainstream changes to be sure and update the repo.

abdullahibnnadjo, 2017-12-11:
Hi, I'm running on `4.14.3-300.fc27.x86_64 GNU/Linux` and had no problem so far with current version.

free5lot, 2017-12-12:
Thanks @abdullahibnnadjo for information, I thought the same.
At the holidays in January I'll try to make a version based on latest kernel code checking kernel code guidelines and will try to report it as a patch to mainstream one more time.

Ndolam, 2017-12-14:
While I have your attention... 
The current version of the code doesn't allow you to swap both Opt/Cmd and RightAlt/RightCtl.
If you are interested in my changes to allow that, let me know and I'll pass them on.
Thanks.

p-himik, 2019-03-20:
I had to manually merge the changes from https://github.com/torvalds/linux/commit/ee345492437043a79db058a3d4f029ebcb52089a in order to make this module work with my keyboard.

free5lot, 2019-03-20:
@p-himik Thanks for information. What is your kernel version and GNU+Linux distribution name/version? And what is your keyboard? 
I should probably update the repo up to date with the kernel and make one more attempt to push the patch to mainstream.

p-himik, 2019-03-20:
@free5lot Ubuntu 18.10, 4.18.0-16-generic. Apple Magic Keyboard (MRMH2LL/A), connected via USB - didn't even try with Bluetooth.
Also, just FYI: I made it possible to swap Fn and EjectCD (IMO the most useless button in our time). If that's useful enough, maybe it's worth to have it before pushing the patch to mainstream.

free5lot, 2019-03-20:
@p-himik good that the patched worked for you. You can make a pull request if you want. Preferable following the kernel's coding rules.
The patched version supports ejectcd_as_delete option, have you tried it? I'm asking because I don't have ejectcd key on my device. And yes, EjectCD is pretty useless nowadays, I agree.

p-himik, 2019-03-20:
@free5lot I can try it, sure. And I'll try to come up with a PR some time soon. I'll probably try to make it more generic to allow exchanging Fn with any other key, as there is already a PR that exchanges Fn and F13, and I can definitely see people want to also use F14-F19.

free5lot, 2020-03-31:
Currently the patched version is up to date with 5.6 kernel and this issue is obsolete and will be closed. If one has different opinion, please write message for reopening.

@p-himik as for your experiments with swapping fn and ejectcd keys: please make a separate issue or PR for this as the current issue has obviously different topic. I'm not sure how demanded it will be as I don't have ejectcd button, but maybe other people will be interested and you will have useful feedback there. Thanks.

closed 2020-03-31 by free5lot