free5lot/hid-apple-patched #43: It stopped working after a few times of suspend and reboot

libreoscar, 2017-12-15:
I was so excited that it worked after installation using the DKMS way. But after a few times of suspend and reboot it stopped working - the keys recovered to their original status... Now rebooting the system does not change anything.

$ uname -a
Linux ThinkCentre-M910t 4.10.0-38-generic #42~16.04.1-Ubuntu SMP Tue Oct 10 16:32:20 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

The keyboard is Apple Magic 2.

libreoscar, 2017-12-15:
$ dkms status
bbswitch, 0.8, 4.10.0-38-generic, x86_64: installedError! Could not locate dkms.conf file.
File:  does not exist.

abdullahibnnadjo, 2017-12-15:
Hi, try to run `modinfo hid_apple | grep parm`. If it prints only a few options (instead of the [6 available with the patch](https://github.com/free5lot/hid-apple-patched#configuration)), then it means that the default `hid_apple` mod was loaded at boot instead of the patch.

You can try to force loading the patch with `modprobe -r hid_apple && modprobe hid_apple` (run the whole in one shot) and see if it works.

I have a little script to do that at boot:
- Systemd service:
```ini
[Unit]
Description=Activate the patch for the Apple keyboard to be able to remap Fn keys etc.

[Service]
Type=oneshot
ExecStart=/usr/local/bin/patch-hid-apple.sh

[Install]
WantedBy=multi-user.target
```
- The script:
```sh
#!/bin/sh
modprobe -r hid_apple && modprobe hid_apple
```

libreoscar, 2017-12-15:
Thank you for the reply. I just tried it out. Seems not working.

$ modinfo hid_apple | grep parm
```
parm:           fnmode:Mode of fn key on Apple keyboards (0 = disabled, [1] = fkeyslast, 2 = fkeysfirst) (uint)
parm:           iso_layout:Enable/Disable hardcoded ISO-layout of the keyboard. (0 = disabled, [1] = enabled) (uint)
parm:           swap_opt_cmd:Swap the Option ("Alt") and Command ("Flag") keys. (For people who want to keep Windows PC keyboard muscle memory. [0] = as-is, Mac layout. 1 = swapped, Windows layout.) (uint)
parm:           swap_fn_leftctrl:Swap the Fn and left Control keys. (For people who want to keep PC keyboard muscle memory. [0] = as-is, Mac layout, 1 = swapped, PC layout) (uint)
parm:           rightalt_as_rightctrl:Use the right Alt key as a right Ctrl key. [0] = as-is, Mac layout. 1 = Right Alt is right Ctrl (uint)
parm:           ejectcd_as_delete:Use Eject-CD key as Delete key. ([0] = disabled, 1 = enabled) (uint)
```

$ sudo modprobe -r hid_apple && modprobe hid_apple
```
modprobe: ERROR: could not insert 'hid_apple': Operation not permitted
```

abdullahibnnadjo, 2017-12-15:
I think you forgot `sudo` for the second command 👍

libreoscar, 2017-12-15:
$ sudo modprobe -r hid_apple; sudo modprobe hid_apple
This one got no error. But nothing happened. Keys are not swapped.

abdullahibnnadjo, 2017-12-15:
I see that all options except `fnmode` and `iso_layout` are set on `[0]` (disabled), do you configured them in `/etc/modprobe.d/hid_apple.conf`?

abdullahibnnadjo, 2017-12-15:
Or you can load them dynamically by appending them to the last command: `modprobe -r hid_apple && modprobe hid_apple swap_fn_leftctrl=1 swap_opt_cmd=1 `... etc

libreoscar, 2017-12-15:
/etc/modprobe.d/hid_apple.conf:
```
options hid_apple fnmode=1
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=0
options hid_apple ejectcd_as_delete=1
```

Tried this command with no luck :-(
`modprobe -r hid_apple && modprobe hid_apple swap_fn_leftctrl=1 swap_opt_cmd=1`

libreoscar, 2017-12-15:
> I see that all options except fnmode and iso_layout are set on [0]

Where did you see it?

abdullahibnnadjo, 2017-12-15:
On the modinfo output, in the description you can see which value is actually set (the one with [])

Oh, just remember about that, last resort, you can modify the parameters directly in `/sys/module/hid_apple/parameters/`, so something like `echo 1 | sudo tee /sys/module/hid_apple/parameters/swap_fn_leftctrl` _**should**_ work

libreoscar, 2017-12-15:
The value of `/sys/module/hid_apple/parameters/swap_fn_leftctrl`.

> On the modinfo output, in the description you can see which value is actually set (the one with [])
I think the one in [.] is the default value. Not the one chosen.

abdullahibnnadjo, 2017-12-16:
Ah you're right, that's what I thought when resolving my issue when installing the patch.
So is it working now?

libreoscar, 2017-12-16:
No. The values in `/sys/module/hid_apple/parameters` are correct, but not taking effect :-(

free5lot, 2017-12-16:
So, the kernel module is loaded instead of original and you did run something like
`echo "1"  |  sudo tee "/sys/module/hid_apple/parameters/swap_fn_leftctrl"` ?
That would be strange, never had such situation on Ubuntu. You may try to reboot and reinstall this module.

free5lot, 2017-12-16:
But the way, can it be that you have autoupdates on and during this reboots and suspends something was auto-installed? You can see it in apt logs probably.

duleorlovic, 2018-07-31:
Is there any way to debug (for example with `printf`) to see if this is being used in the first place ? I'm concerned why fn key is not even detected with `xev`.

I also have Apple Magic Keyboard like @libreoscar 
~~~
dmesg
[  503.780248] usb 1-1.6: New USB device found, idVendor=05ac, idProduct=0267
[  503.780254] usb 1-1.6: New USB device strings: Mfr=1, Product=2,
SerialNumber=3
[  503.780257] usb 1-1.6: Product: Magic Keyboard
[  503.780260] usb 1-1.6: Manufacturer: Apple Inc.
[  503.780262] usb 1-1.6: SerialNumber: FNP7474000HJ20Q1J
[  503.788353] hid-generic 0003:05AC:0267.0007: hiddev0,hidraw3: USB HID v1.10
Device [Apple Inc. Magic Keyboard] on usb-0000:00:1a.0-1.6/input0
[  503.790024] input: Apple Inc. Magic Keyboard as
/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.1/0003:05AC:0267.0008/input/input10
[  503.845869] hid-generic 0003:05AC:0267.0008: input,hiddev0,hidraw4: USB HID
v1.10 Keyboard [Apple Inc. Magic Keyboard] on usb-0000:00:1a.0-1.6/input1
[  503.847653] hid-generic 0003:05AC:0267.0009: hiddev0,hidraw5: USB HID v1.10
Device [Apple Inc. Magic Keyboard] on usb-0000:00:1a.0-1.6/input2
~~~

I'm on Ubuntu 16

~~~
uname -a
Linux main 4.4.0-131-generic #157-Ubuntu SMP Thu Jul 12 15:51:36 UTC 2018 x86_64
x86_64 x86_64 GNU/Linux

lsb_release -r
Release:16.04
~~~

I tried both `./install` and dkms install procedure. When I try third option (usind `make`) than some error occurs:

~~~
sudo make install
make -C /usr/src/linux-headers-4.4.0-131-generic M=/home/orlovic/Programs/hid-apple-patched modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-131-generic'
  INSTALL /home/orlovic/Programs/hid-apple-patched/hid-apple.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-131-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-131-generic'
~~~

I noticed that regardless of the values in `/etc/modprobe.d/hid_apple.conf`
~~~
cat /etc/modprobe.d/hid_apple.conf 
options hid_apple fnmode=2
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1
~~~

when I reboot I always see some `default` values (don't know where they come from)

~~~
tail /sys/module/hid_apple/parameters/*
==> /sys/module/hid_apple/parameters/ejectcd_as_delete <==
0

==> /sys/module/hid_apple/parameters/fnmode <==
1

==> /sys/module/hid_apple/parameters/iso_layout <==
1

==> /sys/module/hid_apple/parameters/rightalt_as_rightctrl <==
0

==> /sys/module/hid_apple/parameters/swap_fn_leftctrl <==
0

==> /sys/module/hid_apple/parameters/swap_opt_cmd <==
0
~~~

free5lot, 2018-08-17:
> I'm concerned why fn key is not even detected with xev.

Because fn in driver's code is considered as a not-real key, modifier, to make fn+F* work correctly. I never  got fn in xev too.

About debugging - it's not easy, and I did not do it. You should look for information why kernel module (in general) does not get parameters from config. And you can try to do the same with the original hid_apple, it has several parameters too.

duleorlovic, 2018-09-20:
Does anybody know how it is solved on Mac ? When I attach Apple keyboard with usb to my Mac and go to Settings -> Keyboard -> Modifier Keys -> Function (Fn) Key and select "Control" instead of "Function", than I can see output in `xev` on my Mac

This if for CTRL keydown and keyup
~~~
KeyPress event, serial 27, synthetic NO, window 0x600001,
    root 0x111, subw 0x0, time 4155578010, (598,463), root:(598,486),
    state 0x0, keycode 67 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 30, synthetic NO, window 0x600001,
    root 0x111, subw 0x0, time 4155578137, (598,463), root:(598,486),
    state 0x4, keycode 67 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
~~~

This is for "fn" keydown and keyup (it is like right Ctrl)
~~~
KeyPress event, serial 30, synthetic NO, window 0x600001,
    root 0x111, subw 0x0, time 4155596282, (598,463), root:(598,486),
    state 0x0, keycode 70 (keysym 0xffe4, Control_R), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 30, synthetic NO, window 0x600001,
    root 0x111, subw 0x0, time 4155596401, (598,463), root:(598,486),
    state 0x4, keycode 70 (keysym 0xffe4, Control_R), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
~~~

free5lot, 2020-05-15:
Now the patched version is up-to-date with the kernel sources and supports external keyboards (both wired and wireless). 
About 2 years passed since the last issue activity and I'm closing it for now but if one has similar installation issue - feel free to add comments and the issue will be reopened.

closed 2020-05-15 by free5lot