free5lot/hid-apple-patched #44: Swapping fn key & left Ctrl not working on Magic Keyboard (Apple Keyboard 2)

jonnyasmar, 2017-12-21:
Title says it all...

This patch doesn't seem to work with the new Magic Keyboard. The switch from the command key to Super seems to work fine from `hid-apple`.

My config looks like this:

```
options hid_apple fnmode=1
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1
```

jonnyasmar, 2017-12-22:
Possible, I suppose - though unlike the OP in https://github.com/free5lot/hid-apple-patched/issues/43 this patched driver never worked for me. And my mistake -- the command/super swap didn't work. I was on way too little sleep when I opened this issue :)

jonnyasmar, 2017-12-22:
Hmm, so I decided to try totally removing everything.

Removed hid-apple, hid-apple-patched, all relevant files. Started from scratch installing ONLY hid-apple-patched (using DKSM),

Tested, rebooted, no luck.

Logged out and booted into Windows to play a few games (no special apple keyboard drivers installed there or anything) and then rebooted back into Ubuntu to get back to work and bam -- keys are swapped! :)

jonnyasmar, 2017-12-22:
A little more information on this: I just plugged the keyboard in this morning when I decided to test it out again and it's been plugged in ever since. Realizing that was the one differentiating factor between today and yesterday, I decided to try unplugging it. Sure enough, the fn & ctrl key swap stopped working. Plugged the keyboard back in and right away, they were back to where they needed to be.

free5lot, 2018-01-11:
So, you say, that when the fn-ctrl swap works for your external Apple Keyboard you can unplug it from USB and plug in back and the swap won't work, is it right?

jonnyasmar, 2018-01-12:
Not exactly @free5lot 

When I unplug the keyboard, the fn swap stops working, but works when it's plugged in. So, it doesn't work at all for the keyboard when connected via bluetooth.

free5lot, 2018-01-12:
Or, I see, you use this keyboard in two ways - connected via Bluetooth and connected directly via USB, don't you? 
And the swap doesn't work at all via bluetooth but works via USB but not always?

Some kind of difficult for me to understand, maybe it will be more clear if you write it out by steps:
0) Booted GNU/Linux with disconnected Magic Keyboard
1) Connected via USB, bluetooth is off - works
2) Connected via bluetooth it's connected via USB - doesn't work.
3) Disconnected blueooth, USB is sill connected - ...
4) Disconnected USB, reconnected via bluetooth - ...
and so on?

jonnyasmar, 2018-01-12:
I do -- partly because it charges while connected, but mostly because I can't get the fn swap to work on bluetooth :(

At this point after doing a complete reinstall of hid-apple-patched, I've been able to get it to work connected via USB 100% of the time, but not at all via Bluetooth. So:

1. Connected via USB, bluetooth off -- works
2. Connected via bluetooth, USB still plugged in -- continues to work (I think USB is prioritized)
3. Disconnected bluetooth, USB still connected -- continues to work
4. Disconnected USB, reconnected via bluetooth -- doesn't work
5. Connected via USB, bluetooth still connected -- works

Hope that helps, but let me know if you need anything!

free5lot, 2018-01-12:
Now the behavior is clear, thanks. The probable reason is that the hid-apple module is not used for connections via bluetooth (and thus the patched code of module), but I'm not sure how it works for external bluetooth keyboards. 
So, guys any ideas are welcome.

free5lot, 2018-01-12:
I have this idea, @jonnyasmar please check the behavior for fnmode and especially swap_opt_cmd flag. They are flags of original hid-apple from the mainstream and the patch doesn't change theirs code at all. So if they work than we might have a bug/issue in patched version, if they don't - the issue is somewhere else, like a general Linux issue.

tonkonogov, 2018-01-22:
Probably, the root cause is a modified code of the FN button in the wireless mode. See the thread here - https://bugzilla.kernel.org/show_bug.cgi?id=99881. There is a patch in the thread as well.

ywpkwon, 2019-01-02:
Hi, I've been using an old Apple keyboard (USB) well with this thankful repo, but recently I got Apple magic keyboard 2 (Bluethooth) and I had similar behavior.  

So I update `hid-apple.c` and `hid-ids.h` by referring to the given thread above. So, I would like to re-install apple-hid/1.0. However, it says

```
sudo dkms add .                                           
Error! DKMS tree already contains: hid-apple-1.0
```

How can I apply the modified `hid-apple.c` and `hid-ids.h` ?

free5lot, 2019-01-16:
@ywpkwon have you tried  swap_opt_cmd flag from original hid-apple? Does it work?

ywpkwon, 2019-01-16:
@free5lot Deeply thanks for your response. Here is my situation. I have two Bluetooth keyboards (A1644 small keyboard, A1843 full keyboard). Although these are Bluetooth, I am just using them with USB connection (somehow Bluetooth pairing doesn't work well, and I'm okay with cables).

A1644 have worked very well with the original hid-apple. Recently, I got A1843 and realized swapping doesn't work this keyboard. 

** I think I had an even older apple keyboard, and it, of course, worked well with the original hid-apple. Only A1843 is questionable. Maybe because it's newer one.

So, I was going to apply those patch, but I don't know how I can apply .patch files.

free5lot, 2019-01-16:
@ywpkwon .patch files? You should probably install module using DKMS, like described in `README.md`. 

But the question is - if you have this new A1843, can you try `swap_opt_cmd` flag with the original (not patched) version of hid-apple, that is already installed in Linux kernel of your system, does this flag work and does it swap keys?
If it does not, then there is no hope for patched hid-apple module to change it, because patched version of hid-apple only extends features, but work in the same way.

free5lot, 2019-01-16:
>I don't need to install DKMS after connecting A1843, right?
That's right.

>Then, is there no hope in using A1843?
Well, probably there is a way to fix it, but some Pull Request is needed, especially from guy who has this not-working devices to be sure the fix is working properly.

ywpkwon, 2019-01-16:
@free5lot Thanks for all you responses. Lastly, may I ask this? Let's say I somehow modified hid-apple.c. How can I update DMS module?

In the intial install I did below.
```
sudo dkms add .
sudo dkms build hid-apple/1.0
sudo dkms install hid-apple/1.0
```
To update, should I remove dkms and re-do the above?
Or, can I do the second `build` command? (I think `add` will be rejected due to existence..?)
I am not sure if this is a question to be answerd simply though.

free5lot, 2019-01-25:
I'm not sure about it and no Macbook around me right now to check, so no advice here.

patkoscsaba, 2019-12-10:
I have a 2012 MacBook Pro. Everything works except fn<->ctrl swap. I see no error on module load, it just simply doesn't apply. All the other options work as expected. Any ideas what could I try? Does nmode=2 affect swap_fn_leftctrl=1 ?

jbriales, 2019-12-16:
@patkoscsaba I came here for the same reason. Everything works (e.g. command-option swap) except for fn-leftctrl swap.
Interestingly it worked for me in a Thinkpad, then changed to another Thinkpad (same model afaict) and with the same setup it doesn't work here. Very strange and frustating!
Give a shout if you find anything else about this scenario.

jbriales, 2019-12-21:
Ok following up my previous message: In my case the installation wasn't succeeding.
Make sure all steps run with 0 exit or it won't work. Ultimately, after everything is installed, I run `sudo modprobe -r hid_apple; sudo modprobe hid_apple` to ensure the new module is used (alternative is reboot).

jbriales, 2019-12-21:
@jonnyasmar did you need to do any special step for the bluetooth to work? I couldn't get the laptop to connect to the keyboard via bluetooth, ever (it isn't listed) :/

free5lot, 2020-03-31:
So, guys, according to @almson who has Apple Magic Keyboard 2 it should be working on GNU+Linux in both modes - wired and bluetooth.
Can you report it's not working for you?
If it is working, the issue will be closed as solved.

free5lot, 2020-05-15:
>  If it is working, the issue will be closed as solved.

So, we may consider it working as we have positive feedback from other Magic Keyboard users.

closed 2020-05-15 by free5lot