free5lot/hid-apple-patched #45: [Ubuntu 17.10] dkms didn't work, but build/install scripts did

derekderaps, 2018-01-09:
FYI, installation via dkms did not work for me (`modinfo hid_apple | grep parm` returned just the three standard options, not the full six of your module) on my mid-2015 MBP. I tried first from wayland, then from xorg, just for kicks, with the same result. But then I tried your build/install scripts, and boom, it works in all the places! Thank you so much for your work on this project; you get special angel points.

free5lot, 2018-01-10:
Thank you Derek for information about  Ubuntu 17.10 and thanks for good words. I'll take a look when I have time. Glad that at least one way of module installation worked for you.

bradlishman, 2018-03-27:
Worked for me too on 17.10 with a1502.

free5lot, 2018-03-30:
@bradlishman worked for you via dkms or build/install script?

bradlishman, 2018-04-08:
@free5lot via the build/install script.

anom, 2019-11-20:
@FastThenLeft  were you able to ever swap teh Fn and Ctrl key?

free5lot, 2020-03-31:
Closing this issue as there are no complains currently about installation of patched module. Also I tested dkms way recently on Ubuntu myself and it works fine.