free5lot/hid-apple-patched #46: CTRL+SHIFT+button combinations dont work

leobunga, 2018-01-21:
First of all thanks a lot for the package!
When working in the terminal I noticed that combinations involving ctrl+shift+t (in my case, for opening a new tab) wouldnt work. single ctrl or shift combinations work fine, as well as everything else so far.

free5lot, 2018-01-22:
I bet you have an external keyboard, not a notebook like MacBook Pro?
It's a known issue, that is probably unsolvable for external keyboards (on Macbook Pro it works).
https://github.com/free5lot/hid-apple-patched/issues/6
It seems, that this exact combination is ignored on even lower level, not sure why it and why others not.

You can follow [this link](https://github.com/free5lot/hid-apple-patched/issues/6#issuecomment-331513188) and check this behavior on original MacOS with the same swap, it would be helpful, because I don't have external keyboard to check this myself.

free5lot, 2018-10-19:
Closed as unsolvable duplicate.

closed 2018-10-19 by free5lot

djdevin, 2019-02-05:
FWIW if you train yourself you might be able to get around this limitation. The keys cannot simultaneously be pressed but you have a small window which may be recognize as simultaneous.

"Roll" up to Control-Shift-T by pressing no more than 2 keys at the same time (control, then shift, then start to pull off as you press T) and you should be able to trick it into thinking you pressed them all together. I accidentally discovered this and I can get it 99% of the time now.

free5lot, 2019-02-06:
@djdevin Wow, interesting to hear, thanks for sharing. I'll repost it to the original issue #6.