free5lot/hid-apple-patched #47: ejectcd_as_delete doesn't work on Macbook 4,1

dafzor, 2018-06-09:
When using hid-apple-patched all options work as expected except using the eject key for delete.
When in apple boot menu the eject key does eject the disc so it is functional.

System:
macbook 4,1
Manjaro with 4.14 LTS kernel (only OS).
https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/ package

free5lot, 2020-03-31:
@dafzor any information on the issue? I mean nobody reported the same but maybe this is Macbook 4,1 specific. Please try up-to-date version as today it was updated with merge of upstream changes.

dafzor, 2020-04-01:
Still no change as far as I can tell. Would there be any test i could do to even check if the key press is even registering as anything?

free5lot, 2020-04-02:
>Would there be any test i could do to even check if the key press is even registering as anything?

You can try `xev` to find out if the key Eject returns any code (better start with original hid-apple to be sure).

I have no CD drive on my Macbook Pro and no Eject button accordingly, but the idea of using eject as delete was taken from @JanmanX 's patch.

Maybe someone who has Eject button and uses it as delete can provide some guidance here?

dafzor, 2020-04-05:
xev doesn't show output using the patched driver, default:
```
KeyPress event, serial 37, synthetic NO, window 0x3400001,
    root 0xd8, subw 0x0, time 186680, (622,349), root:(622,373),
    state 0x0, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x3400001,
    root 0xd8, subw 0x0, time 186760, (622,349), root:(622,373),
    state 0x0, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 40, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248
```

LilithSilver, 2024-08-04:
I'm having the same issue on MacBook Pro 9,2 (mid-2012 13-inch). Everything else works OK, but `ejectcd_as_delete` doesn't show up in `/sys/module/hid_apple/parameters` at all. I'm unable to write a test file to `parameters` like the readme says to see if that changes anything at runtime, since it's a virtual filesystem.

`xev` shows no output when eject is pressed. The key is functional and does indeed perform an eject operation. GNOME is also able to detect the eject and show an icon for it. 

Using a fresh install of 6.10.2-arch1-1, and installed via the Arch setup script.

LilithSilver, 2024-08-04:
Here's a configuration workaround. Follow [these instructions](https://askubuntu.com/a/1019659).

My configuration file ended up looking like this:

```
/etc/udev/hwdb.d/65-keyboard-custom.hwdb                  
evdev:input:b0003v05ACp0252e0111*
    KEYBOARD_KEY_C00B8=delete  # map Eject to Delete
```

Follow the instructions in the linked answer to get your hardware info; it might be slightly different. Note that the vendor ID and keycode letters must be capitalized.