free5lot/hid-apple-patched #48: Support for bluetooth

nmattia, 2018-09-02:
This worked out of the box for my magic keyboard 2, thanks @free5lot ! 

I'm having some issues over bluetooth, I think the wrong driver is being picked up. It might default to `hid-generic` by default. If I specify the driver in my xorg.conf:

```
Section "InputClass"
        Identifier "magic keyboard"
        Driver "hid_apple"
        MatchProduct "Magic"
        MatchIsKeyboard "on"
EndSection
```

the keyboard becomes completely unusable (e.g. no key works). I noticed that the `hid-ids.h` specifies an ID for the keyboard (0x0267) which is the one I get from `lsusb` when I connect the keyboard through usb. Could it be that the kernel needs to know about the keyboard through some other mechanism over bluetooth?

free5lot, 2018-09-19:
> This worked out of the box for my magic keyboard 2, thanks @free5lot !

You are welcome!

About bluetooth connectivity - I don't know actually how to do this and have no bluetooth keyboard (especially apple one) to check.

But I'll leave this issue opened if you or someone other find any useful information and post it here, I'll add it to the readme.

Aetf, 2019-07-30:
Related [comments](https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/#comment-702311) from [jmt](https://aur.archlinux.org/account/jmt) in AUR:
> Has anyone else had issues with this on Linux 5.x? After I upgraded my kernel to 5.2, whenever I connected my Apple Magic keyboard over bluetooth, the hid-generic driver was being used and not the hid_apple module. Once I uninstalled this package the correct driver was being used. I wonder if this patch has a conflict with this package?
>
> https://lore.kernel.org/patchwork/patch/978072/

daniellockyer, 2020-01-24:
I've found the fix that needs to be applied to the repo, but I believe it also requires a commit to the kernel to fix a missing definition. I'll see what I can do

almson, 2020-03-30:
Merging the mentioned patch (https://lore.kernel.org/patchwork/patch/978072/) causes the Magic Keyboard to work in Bluetooth mode. Swapping of keys works, but Fn changes the backlight. I'd try to help, but the hid-apple code is such a mess! I'll stick to using a USB cable for now.

almson, 2020-03-30:
Merging upstream fixes bluetooth. See PR #63

free5lot, 2020-03-31:
Nice, than I'm closing this issue.

@nmattia and others interested parties - please check current patched version and in case of any issues feel free to message here or in separate issue is you have any troubles.

closed 2020-03-31 by free5lot