free5lot/hid-apple-patched #49: How do I get my system to use the patched version of hid-apple (Arch)

Hubro, 2019-07-15:
I've installed this patch via AUR (`hid-apple-patched-git-dkms`) (CC @Aetf)

That included this config file (`/etc/depmod.d/hid-apple-patched.conf`):

```
override hid-apple * extra
```

As well as (`/etc/modprobe.d/hid_apple_pclayout.conf`):

```
options hid_apple fnmode=2
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1
```

I ran `sudo depmod -a` and `sudo mkinitcpio -p linux` and rebooted.

Immediately the command and opt keys were swapped, and my F keys worked without having to hold fn, which is great. All the other options were ignored though, my fn key and control keys are not swapped. I can't set these settings under `/proc`, I just get access denied, even though I'm root.

That makes sense because apparently not all options are available:

```
 $ ls /sys/module/hid_apple/parameters/
fnmode  iso_layout  swap_opt_cmd
```

Is this because my system is using the built-in version of hid-apple rather than the patched one? How can I know which one is being used?

(I'm using the wireless magic keyboard via Bluetooth, but I'm guessing that's not relevant.)

Here's the relevant lines from dmesg:

```
[Jul15 10:51] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0.000004] Bluetooth: HIDP socket layer initialized
[  +1.084478] hid-generic 0005:004C:0267.0003: unknown main item tag 0x0
[  +0.000117] input: Magic Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:2/0005:004C:0267.0003/input/input34
[  +0.000382] input: Magic Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:2/0005:004C:0267.0003/input/input35
[  +0.000268] hid-generic 0005:004C:0267.0003: input,hidraw2: BLUETOOTH HID v1.00 Keyboard [Magic Keyboard] on 9c:b6:d0:e7:e9:4c
[  +0.006218] hid_apple: unknown parameter 'swap_fn_leftctrl' ignored
[  +0.000003] hid_apple: unknown parameter 'rightalt_as_rightctrl' ignored
[  +0.000003] hid_apple: unknown parameter 'ejectcd_as_delete' ignored
[  +0.146013] apple 0005:004C:0267.0003: unknown main item tag 0x0
[  +0.000135] input: Magic Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:2/0005:004C:0267.0003/input/input36
[  +0.000524] apple 0005:004C:0267.0003: input,hidraw2: BLUETOOTH HID v1.00 Keyboard [Magic Keyboard] on 9c:b6:d0:e7:e9:4c
```

Some system and version info:

```
 $ pacman -Qs hid
local/hid-apple-patched-git-dkms 20171118.c92cc98-1
    Allows to swap the Fn key and left Control key on Macbook Pro and Apple keyboards in GNU/Linux (DKMS)

 $ uname -a
Linux carbon 5.1.16-arch1-1-ARCH #1 SMP PREEMPT Wed Jul 3 20:23:07 UTC 2019 x86_64 GNU/Linux

 $ cat /etc/*-release
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="0;36"
HOME_URL="https://www.archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux
```

Aetf, 2019-07-16:
You are still using the builtin one. You will see all options under `/sys/module/hid_apple/parameters` if the patched one is being used.

You can check `dkms status` and see if the module is correctlt compiled.

free5lot, 2019-07-17:
Yes, @Aetf is right, you don't have patched version running. The options you have (`fnmode`, `iso_layout`, `swap_opt_cmd`) are from the original version. Patched version adds additional ones like `swap_fn_leftctrl`.
So, try to find out why you don't have new version of kernel module loaded.

Hubro, 2019-07-19:
> You are still using the builtin one. You will see all options under `/sys/module/hid_apple/parameters` if the patched one is being used.
> 
> You can check `dkms status` and see if the module is correctly compiled.

Thanks for the response. I can't seem to get `dkms status` to produce any output:

```
 $ dkms status
 $ sudo dkms status
 $ sudo dkms status hid-apple
 $ sudo dkms status hid-apple-patched
```

Aetf, 2019-07-20:
Do you have `dkms` package correctly installed? Your kernel isn't the latest one. Probably you should do a full system upgrade first.

How did you install `hid-apple-patched-git-dkms`? Did it run successfully?
There should be dkms hook triggered by installing the package, and you should see outputs like by the end of the installation.
```
==> dkms install hid-apple-patched/20170406.61dce7d -k 5.2.1-arch1-1-ARCH
```

Maybe you can try reinstall these two packages. Also see if there's anything in the Archlinux [wiki](https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support) will help.

Hubro, 2019-07-20:
@Aetf Thanks, I figured it out. When I installed the package from AUR, it printed a line saying "missing kernel headers". I installed `linux-headers` and the DKMS hook immediately installed hid-apple-patched.

After rebooting, my Magic Keyboard is now handled by hid-generic... All of the improvements of the non-patched hid_apple disappeared.

Manually activating `hid-apple` by running `sudo modprobe hid_apple`, I now see all parameters available:

```
 $ ls /sys/module/hid_apple/parameters/
ejectcd_as_delete  fnmode  iso_layout  rightalt_as_rightctrl  swap_fn_leftctrl  swap_opt_cmd
```

But none of the parameters affect my keyboard in any way. Is there any way to force `hid_apple` to handle my keyboard?

Here's my dmesg output now, when connecting the keyboard:

```
[  +8.709086] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0.000004] Bluetooth: HIDP socket layer initialized
[  +1.359279] hid-generic 0005:004C:0267.0003: unknown main item tag 0x0
[  +0.000180] input: Magic Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:2/0005:004C:0267.0003/input/input31
[  +0.000357] input: Magic Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:2/0005:004C:0267.0003/input/input32
[  +0.000140] hid-generic 0005:004C:0267.0003: input,hidraw2: BLUETOOTH HID v1.00 Keyboard [Magic Keyboard] on 9c:b6:d0:e7:e9:4c
```

-----

I did a full system update while I was at it.

```
 $ uname -a
Linux carbon 5.2.1-arch1-1-ARCH #1 SMP PREEMPT Sun Jul 14 14:52:52 UTC 2019 x86_64 GNU/Linux
 $ pacman -Qs apple
local/hid-apple-patched-git-dkms 20171118.c92cc98-1
```

Aetf, 2019-07-20:
Hmm, I'm not an expert regarding this, but seems you can at least manually rebind the keyboard to use hid-apple like [this](https://unix.stackexchange.com/q/240573/221101).

Hubro, 2019-07-20:
@Aetf Disappointment... :(

I could unbind successfully from hid_generic, but I couldn't bind to hid_apple, I just get an error:

```
 $ echo -n 0005:004C:0267.0003 | sudo tee /sys/bus/hid/drivers/apple/bind 
0005:004C:0267.0003tee: /sys/bus/hid/drivers/apple/bind: No such device
```

I also attempted the X config from issue #48, but it didn't affect my keyboard in any way, it still uses hid_generic.

So doesn't the patched driver support Bluetooth at all? Shouldn't it support it, considering the built-in hid_apple supports Bluetooth?

almson, 2020-03-30:
You are connecting over Bluetooth. This currently doesn't work. You can use a USB cable.

free5lot, 2020-03-31:
@Hubro not it should support (see #48, thanks to @almson with their PR #63).
Please check it with your device.

free5lot, 2020-03-31:
@Aetf as a maintainer of https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/ could you please check current version and update AUR?

Aetf, 2020-03-31:
Done. https://github.com/free5lot/hid-apple-patched/issues/57#issuecomment-606776408

free5lot, 2020-04-01:
Thank you @Aetf. 

This issue will be closed as solved soon if @Hubro is not going to provide any additional feedback.

Hubro, 2020-04-01:
> Thank you @Aetf.
> 
> This issue will be closed as solved soon if @Hubro is not going to provide any additional feedback.

Well it seems the conclusion is that this patched driver doesn't support Bluetooth, so I'm not sure there's any value in keeping this issue open :) Unless anyone intends to implement it.

almson, 2020-04-01:
@Hubro The patched driver does support Bluetooth now.

free5lot, 2020-04-01:
>Well it seems the conclusion is that this patched driver doesn't support Bluetooth, so I'm not sure there's any value in keeping this issue open :) Unless anyone intends to implement it.

No, it supports now. That's the point of bumping this ticket (reread my recent messages here). Please check the new version and if everything's fine the issue will be closed as solved.

Hubro, 2020-04-01:
@almson @free5lot My bad, I'll give it a try as soon as I have time.

Hubro, 2020-04-09:
@free5lot Had some time to test this today. I installed hid-apple-patched-git-dkms from AUR on kernel 5.6.3 and everything seems to work perfectly out of the box :) Command and alt is swapped, Ctrl and Fn is swapped, all F keys are F keys by default and the media keys work as expected by using the Ctrl key.

Awesome job!

closed 2020-04-09 by Hubro

free5lot, 2020-04-10:
@Hubro nice it works for you.

>Except one small thing, the key to the right of my shift key is swapped with the key above my tab key. On my keyboard layout that means the < key and | keys are swapped. That's not relevant to this patch though.

Try iso_layout option like that:
echo 1 | sudo tee /sys/module/hid_apple/parameters/swap_fn_leftctrl
or
echo 0 | sudo tee /sys/module/hid_apple/parameters/swap_fn_leftctrl

If it fixes your issue you will be able to set this setting permanently in config.

Hubro, 2020-04-14:
> @Hubro nice it works for you.
> 
> > Except one small thing, the key to the right of my shift key is swapped with the key above my tab key. On my keyboard layout that means the < key and | keys are swapped. That's not relevant to this patch though.
> 
> Try to change `iso_layout` option like that:
> `echo 1 | sudo tee /sys/module/hid_apple/parameters/iso_layout`
> or
> `echo 0 | sudo tee /sys/module/hid_apple/parameters/iso_layout`
> 
> If it fixes your issue you will be able to set this setting permanently in config.

That doesn't make any difference unfortunately :frowning_face:

p-himik, 2024-04-01:
I'm now having this problem as well. The module is indeed installed, the kernel has loaded it - all the options are there.
It's magically fixed by `sudo modprobe -r hid_apple; sudo modprobe hid_apple`.

`usb-devices` shows that the keyboard uses `usbhid`. I don't know whether it's relevant or not - none of the values change after the above command.
```
T:  Bus=01 Lev=05 Prnt=12 Port=02 Cnt=01 Dev#= 14 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05ac ProdID=026c Rev=08.52
S:  Manufacturer=Apple Inc.
S:  Product=Magic Keyboard with Numeric Keypad
S:  SerialNumber=F0T85060145HTCYA1
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=8ms
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
E:  Ad=82(I) Atr=03(Int.) MxPS=  16 Ivl=8ms
```

`dkms status` shows:
```
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/hid-apple/1.0/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/hid-apple/1.0/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/hid-apple/1.0/source/dkms.conf)
hid-apple/1.0, 6.5.0-21-generic, x86_64: installed
hid-apple/1.0, 6.5.0-25-generic, x86_64: installed
hid-apple/1.0, 6.5.0-26-generic, x86_64: installed
```