free5lot/hid-apple-patched #52: ./install

anom, 2019-10-20:
I'm getting the following error whilst running ./install.sh

    rmmod: ERROR: Module hid_apple is not currently loaded
    insmod: ERROR: could not insert module ./hid-apple.ko: Operation not permitted

Any suggestions?

Ndolam, 2019-10-20:
On Sun, Oct 20, 2019 at 02:25 (-0700), anom wrote:

> I'm getting the following error whilst running ./install.sh

> rmmod: ERROR: Module hid_apple is not currently loaded
> insmod: ERROR: could not insert module ./hid-apple.ko: Operation not permitted

> Any suggestions?

Try running th command as root, not an ordinary user.

anom, 2019-10-21:
Yeah, tried running as sudo but same error.

free5lot, 2019-10-21:
Hi @anom!
1) obvious recommendation would be run `sudo -i` before command, but may not help.
2) maybe the problem and solution are mentioned here: [[SOLVED] insmod error "operation not permitted" as root](https://www.linuxquestions.org/questions/linux-software-2/insmod-error-operation-not-permitted-as-root-836570/)
3) just for your information - install.sh according to documentation is not preferable way to install this patched module/

free5lot, 2019-10-21:
UPD. Updated link in the previous message.

free5lot, 2020-03-31:
@anom  any update on this?
BTW just recently Bluetooth support was added (migrated from upstream) and Apple Magic 2 keyboard was reported to work properly with this patched version for both Bluetooth and USB modes. So please try again or this issue will be closed.

free5lot, 2020-05-15:
So, we may consider it working as we have positive feedback from other users and I've tested it today once more myself.

closed 2020-05-15 by free5lot