free5lot/hid-apple-patched #53: Swap left ctrl and left cmd

willir, 2019-12-04:
Could you please and the option to swap left ctrl and left cmd buttons?

Currently, I have to do via KDE settings, and change it every time I switch to a regular keyboard.

free5lot, 2019-12-22:
Well, it's kind of stuff that is possible to do in a usual way (as you do via KDE settings).
The patched version targets FN button mostly, because it's impossible to swap it in any other easy way.
The issue will stay open for a while, maybe some one will express opinion on this.

almson, 2020-03-30:
@willir Why do you have to change settings when you switch to a regular keyboard? Install this module with PC-like settings, and then you can use the same key remapping settings for Apple and non-Apple keyboards.

I agree that setxkbmap already supports swapping everything except Fn. The number of swap combinations is endless, and it's actually tricky to implement, at least without rewriting the module.

willir, 2020-03-30:
@almson  I'm not quite sure what you proposing. I don't swap left ctrl and left win on regular keyboards, I only need it on Apple one.

So on a regular one, I would like to have a regular mapping.

Can I set up a mapping via `setxkbmap` for each keyboard individually?

Thanks for your reply!

almson, 2020-03-30:
@willir I don't understand. If you swap left control and left win on an Apple keyboard, it won't match the keys of a regular keyboard.

willir, 2020-03-30:
@almson  Why wouldn't it?  
Well, it seems it does. Maybe I do it the wrong way.

I map keys via KDE settings:

[image]

When I attach a regular keyboard, it has win and ctrl swapped.

almson, 2020-03-31:
@willir Well, if you install hid-apple-patched and configure it as recommended, then your Apple keyboard will match a PC keyboard and you won't need to swap any other keys.

willir, 2020-03-31:
@almson You mean to swap fn and ctrl? So ctrl is the leftmost one, and fn is to the right, like on usual keyboards?  
Yeah, that was always an option, however, at least for me, it is not convenient. The Apple keyboard has relatively small fn, and probably placed inconveniently, so when it is a ctrl my little finger hurts. For the Apple keyboard, I find it to be convenient to place ctrl on cmd. 

I understand that it might not be for everybody, but that what I find convenient. And it is only for the Apple keyboard, for regular keyboards I like regular mapping.

I also understand that if nobody else, except me, needs it, the change might not worth extra logic it requires. Since extra logic, for example, might lead to more bugs.

almson, 2020-03-31:
@willir I suggest to set the hid-apple driver to the regular PC layout, and then find an option that you like. Then, you'll have the same behavior on all of your keyboards. If you really want different keyboards to have different layouts, then you should be able to do it with Xkb. See https://wiki.archlinux.org/index.php/X_keyboard_extension

willir, 2020-03-31:
@almson Thanks for the links, and suggestions I'll try those out.

I'm closing the issue then.

closed 2020-03-31 by willir