free5lot/hid-apple-patched #54: 5.4 Kernel build issue

dustin-hawkins, 2019-12-13:
```
dhawkins@dhawkins-MacBookAir:~/build/hid-apple-patched$ sudo dkms build hid-apple/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.2-050402-generic -C /lib/modules/5.4.2-050402-generic/build M=/var/lib/dkms/hid-apple/1.0/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.2-050402-generic (x86_64)
Consult /var/lib/dkms/hid-apple/1.0/build/make.log for more information.
```

Having trouble building on a newer 5.4 Kernel - ubuntu 18.04 based. Make.log attached

[make.log](https://github.com/free5lot/hid-apple-patched/files/3961686/make.log)

dustin-hawkins, 2019-12-13:
I think this has to be a version mismatch of GCC.

7.4 on my system, and the newest kernel is compiled w/ the 9.2 branch

almson, 2020-03-30:
@free5lot Issue should be closed. (Issue is with user's system's inability to compile any kernel modules.)

free5lot, 2020-03-30:
@almson agreed. Considering this being solved. If anyone has other opinion - please write a message.

closed 2020-03-30 by free5lot