free5lot/hid-apple-patched #55: Kernel headers not found in Fedora

jbriales, 2019-12-21:
In some of my Fedora computers, when doing `sudo dkms build hid-apple/1.0` I got an error like:
> Error! echo
> Your kernel headers for kernel 5.2.11-200.fc30.x86_64 cannot be found at
> /lib/modules/5.2.11-200.fc30.x86_64/build or /lib/modules/5.2.11-200.fc30.x86_64/source.

Eventually I figured out the issue:
You have to install the kernel-devel package for your specific kernel, like `sudo dnf install kernel-devel-$(uname -r)`.
*However* apparently not all kernel-devel versions are available through the package repository.
So if you get an error upon trying to install `kernel-devel-$(uname -r)` as above,
download concrete kernel-devel package from web https://kojipkgs.fedoraproject.org/packages/kernel/
This can be done directly from URL,
e.g. `sudo dnf install https://kojipkgs.fedoraproject.org//packages/kernel/4.17.9/200.fc28/x86_64/kernel-devel-4.17.9-200.fc28.x86_64.rpm`

Sharing here in case it helps someone else, as it took me a lot of attempts and reading to finally come into this really simple solution :)

free5lot, 2019-12-22:
Thank you for information. I will consider adding it to readme (description) during holidays.

free5lot, 2019-12-22:
The issue will be closed as it's not related to hid-apple itself, but will stay available for further comments and discussions if needed.

closed 2019-12-22 by free5lot