free5lot/hid-apple-patched #57: Step by step instructions for Manjaro linux / Arch linux?

hyperfocus1337, 2020-01-09:
It seems the instructions for Manjaro/Arch are different compared to the Ubuntu instructions in the README of the repo. So far I haven't able to figure out how to get the patch to work. The Arch wiki [Apple Keyboard](https://wiki.archlinux.org/index.php/Apple_Keyboard#Treating_Apple_keyboards_like_regular_keyboards) is useful but difficult to parse for newbies.

I've installed the AUR package `hid-apple-patched-git-dkms`, which already creates the configuration file `/etc/modprobe.d/hid_apple_pclayout.conf` making the instructions in the README unnecessary.

The command `sudo update-initramfs -u` doesn't work in Manjaro/Arch, it seems mkinitcpio is the equivalent utility.

The Arch wiki mentions: `After installation the change is not picked up by the kernel immediately. The simplest way is to just reboot your system and the new behavior should be in effect.` This doesn't seem to be the case for my system.

This is the part of the wiki where I get lost: `Note: Do not forget to include the configuration file in initramfs otherwise it will not work automatically after boot. Refer to Mkinitcpio#BINARIES and FILES or Mkinitcpio#HOOKS (the hook you might need is called).` 

When I open the configuration file `/etc/mkinitcpio.conf`, I can see an empty `BINARIES=()` configuration array. Am I supposed to put something here?

I've also tried running: `sudo modprobe -r hid_apple; sudo modprobe hid_apple` to no avail.

Anyhow, I'm lost, can someone point me in the right direction with step by step instructions?

free5lot, 2020-01-12:
I'm not using Manjaro/Arch, so I hope someone with this specific knowlagde will help you here.

almson, 2020-03-30:
Note that at the moment, the Magic Keyboard 2 only works through USB, not Bluetooth.

The instructions are just to install `hid-apple-patched-git-dkms` from AUR and reboot, provided you installed kernel headers, which you did.

free5lot, 2020-03-31:
@hyperfocus1337 and @almson can you figure out the current state of this issue? I mean AUR package probably should be updated now to include this bluetooth upstream changes but maybe currently installation around AUR is an options?

free5lot, 2020-03-31:
@Aetf as a maintainer of https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/ could you please check current version and update AUR?

almson, 2020-03-31:
Re-installing the AUR package will pull the latest commit (because it's a -git package), but for existing users to see an update the version in AUR needs to be bumped.

Actually, there is also an issue with installing the AUR package where initramfs is not rebuilt. Rebuilding manually can be done with `sudo mkinitcpio -P all`. I'm not sure what PKGBUILD option is needed to do this automatically. @Aetf do you know?

Aetf, 2020-03-31:
There is a packman hook for rebuilding initramfs, but that's only for kernel updates and alike. I think I can just add a notice to the message after installing the package.

Aetf, 2020-03-31:
I updated the package, which now includes a link to the wiki page on how to update `initramfs`. I also updated the wiki page to have more specific instructions.

https://wiki.archlinux.org/index.php/Apple_Keyboard#Including_the_configuration_file_in_initramfs

almson, 2020-03-31:
Let's continue this conversation in #31 
Current issue should be closed as a duplicate. Sorry for posting here.

free5lot, 2020-04-01:
@hyperfocus1337 so, your information and desire for instructions for Arch/Manjaro was acknowledged and may result to adding small instruction block to README. The further feedback is recommended to be provided in the issue #31. If you think that it's not a duplicate or something else about closing this issue - feel free to post here and ask for reopen.

closed 2020-04-01 by free5lot

free5lot, 2020-04-04:
@hyperfocus1337 detailed instruction for installation on Arch Linux was added to arch wiki (related PR #64)

https://wiki.archlinux.org/index.php/Apple_Keyboard#Use_a_patch_to_hid-apple