free5lot/hid-apple-patched #66: control + shift + t doesn't seem to work

ygurin, 2020-05-10:
It seems like control + shift + t doesn't seem to work in chrome (for reopening close tabs) even though control + t works (that opens a new tab). This an issue? Am I missing something...

free5lot, 2020-05-11:
Hi @ygurin, unfortunately it seems to be a hardware limitation of external apple keyboards. The entire 5-T-G-B column is affected by this issue. You can find more information in issues #6 and #46.

And there is a **semi-solution** for this problem - rolling pressing that worked for people with external keyboards, more information is available in #6 and #46. 

I'm closing this issue as a duplicate but feel free to add any replies here of in #6 and #46.

closed 2020-05-11 by free5lot

free5lot, 2020-06-15:
Some not from #6 :
> Just press "Fn+Shift+T", then release the Fn key first. That will always register a Ctrl+Shift+T press.