free5lot/hid-apple-patched #67: Only getting the default options after patch install.

luis-alves, 2020-05-24:
`$ uname -a`
`Linux name 5.6.11-arch1-1 #1 SMP PREEMPT Wed, 06 May 2020 17:32:37 +0000 x86_64 GNU/Linux`
Keyboard: Apple magic keyboard 2 (A1644)

1- I installed the hid-apple-patch through Arch AUR.
2- Ran `sudo mkinitcpio -p linux`
3- Rebooted 
But still only have these three options available:
`$ ls /sys/module/hid_apple/parameters/`
`fnmode  iso_layout  swap_opt_cmd`

In hid_apple.conf there is the correct options:
`cat /etc/modprobe.d/hid_apple.conf  `
`options hid_apple fnmode=2
options hid_apple swap_fn_leftctrl=0
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1`

I don't get any output from `dkms status`.
I'm I missing something?

ozdemir1419, 2020-06-07:
@luis-alves
I bought [ Apple Magic Keyboard 2, without numpad version, Turkish Q Layout ] two days ago.
I installed Arch Linux ( Turkish F Layout ) yesterday.
This module is working fine except two keys. [ + ] and  [ < ] mixed.
### <ins>LOG</ins>
```
# dkms status
hid-apple, 1.0, 5.6.15-arch1-1, x86_64: installed
hid-apple-patched, 20200410.d5ffaff, 5.6.15-arch1-1, x86_64: installed (original_module exists)

# ls /sys/module/hid_apple/parameters/
ejectcd_as_delete  fnmode  iso_layout  rightalt_as_rightctrl  swap_fn_leftctrl	swap_opt_cmd

# uname -a
Linux arch 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000 x86_64 GNU/Linux

# cat /etc/modprobe.d/hid_apple.conf 
options hid_apple fnmode=2
options hid_apple swap_fn_leftctrl=1
options hid_apple swap_opt_cmd=1
options hid_apple rightalt_as_rightctrl=1
options hid_apple ejectcd_as_delete=1
```

### <ins>Here is how I installed</ins>

\# 1 
`pacman -S base-devel dkms linux-headers`

\# 2 This Aur package is dependency.
<a href="https://aur.archlinux.org/packages/mkinitcpio-modconf-hook-git/">aur.archlinux.org/packages/mkinitcpio-modconf-hook-git/</a>

\# 3 I think Aur package is recommended way 
<a href="https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/">aur.archlinux.org/packages/hid-apple-patched-git-dkms/</a>

<hr>

Note: I installed Aur packages with `makepkg -si` command. And I did not run `mkinitcpio -p linux` command. Aur package automaticly updates initramfs.

luis-alves, 2020-06-07:
Thanks @ozdemir1419 . I tried one last thing: change the kernel to lts and got it to work.

free5lot, 2020-06-08:
@luis-alves so we can consider this installation issue to be solved?

vasilakisfil, 2020-06-11:
@luis-alves I also installed that to a new laptop, running 20.04 pop_os:

```
Linux pop-os 5.4.0-7634-generic #38~1590707589~20.04~1dd0040-Ubuntu SMP Fri May 29 01:55:10 UTC  x86_64 x86_64 x86_64 GNU/Linux
```

I can also see only 3 parameters in `/sys/module/hid_apple/parameters/` and needless to say that the ones missing, are not working.

Which kernel did you install and made it work ?

luis-alves, 2020-06-11:
@vasilakisfil my kernel is: 5.4.43-1-MANJARO, don't forget to add kernel headers and restart.
@free5lot Yes the issue is closed by my part. Thanks for this tool and your time!

vasilakisfil, 2020-06-11:
I updated the kernel to:

```
Linux pop-os 5.4.43-050443-generic #202005271238 SMP Wed May 27 16:42:30 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

and rebuild the module but no changes. Still seeing 3 options in parameters. any ideas ?

vasilakisfil, 2020-06-11:
For some reason that I don't know, if I install it through the scripts (build.sh & install.sh) it works. Weird!

free5lot, 2020-06-11:
@vasilakisfil sorry, I have no idea.

>For some reason that I don't know, if I install it through the scripts (build.sh & install.sh) it works.

Now we see that it's installation (so distro-related) problem, not about module itself. Though, I will keep this issue open for some time, maybe somebody will provide useful ideas or you will solve it yourself and describe the details.

free5lot, 2020-08-23:
So what's the best option here if we have one? 
Omitting `DEST_MODULE_LOCATION` will break Arch and Mint, won't it? Setting it to `/kernel/drivers/hid` will make Mint work but break Arch? I don't get what the silver bullet is here.
Maintaining several dkms versions does not look great, does it?

bambinodino, 2020-08-25:
Mint will work with `DEST_MODULE_LOCATION` set to `/extra`. But it will require an extra configuration file to be created by the user or package maintainer. I just tried it myself with the module installed to `/extra`. I created `/etc/depmod.d/hid-apple.conf` as described in https://github.com/free5lot/hid-apple-patched/issues/67#issuecomment-678392903, the way the Arch package maintainer does. The configuration file needs to be there before the module is installed using dkms.

Using `/extra` in conjunction with `/etc/depmod.d/hid-apple.conf` should work with most distributions and their respective packaging policies. So, the documentation needs to be amended and all should work fine, whether the kernel in use comes with its own hid-apple module or not.

free5lot, 2020-08-29:
@bambinodino and others, maybe current `/extra` value is not valid?
I mean, the documentation states `DEST_MODULE_LOCATION` **must** start with `/kernel`.
Maybe something like that should be right?
`DEST_MODULE_LOCATION="/kernel/../extra/"`

@Aetf what do you think as an AUR package maintainer?

free5lot, 2020-08-29:
Well I also checked the history and `DEST_MODULE_LOCATION` was changed from `'/kernel/drivers/hid'` to `'/extra'` 3 years ago to support both Ubuntu (who ignores it) and Arch according to the commit: 
https://github.com/free5lot/hid-apple-patched/commit/e652e4511e6cf0aaeee8ce91e0282f6aee12affa

Not sure why did I do this actually. And why `'/kernel/drivers/hid'` was not working for Arch.
I request help from Arch users here :)

free5lot, 2020-08-29:
I found the source - it was mentioned by @almson here: https://github.com/free5lot/hid-apple-patched/issues/31#issuecomment-294473434

Aetf, 2020-08-29:
I'm also not very knowledgeable of dkms. This package is my first dkms package anyway...

`/kernel/drivers/hid` will not work in Arch packages because packages aren't supposed to modify files owned by other packages (the original hid-apple modules is owned by the `linux` package).

The `/extra` indeed seems a little bit wired, but I've also seen it's been used elsewhere. Anyway, I take a look as a few other packages for their choices of `DEST_MODULE_LOCATION`:

* The Debian wiki uses `/extra` in their example: https://wiki.debian.org/KernelDKMS
* [bcwc_pcie](https://github.com/patjak/bcwc_pcie/blob/master/dkms.conf) module for MBP's webcam also uses `/extra`
* the Arch official (in community repo) [virtualbox-host-modules-dkms](https://github.com/archlinux/svntogit-community/blob/packages/virtualbox/trunk/virtualbox-host-dkms.conf) uses `/kernel/misc`
* However, [virtualbox-host-modules-arch](https://github.com/archlinux/svntogit-community/blob/packages/virtualbox-host-modules-arch/trunk/PKGBUILD#L29), which is a compiled version of `virtualbox-host-modules-dkms` for Arch stock kernel, explicitly places compiled modules in `/extramodules`
* Most packages in extra and community repo in Arch follows this: a `-dkms` version that installs to `/kernel/whatever` and a precompiled version installs to `/extramodules`

Also, we are in a very unique situation that we want to override an existing module of the same name in the kernel, while all the above packages are providing completely new modules.

So at the end of the day, I guess as long as dkms can successfully compile the module to the target location, this shouldn't matter? Because we can always use  `/etc/depmod.d/hid-apple.conf` to specify where to find our version.

free5lot, 2020-08-30:
Thank you @Aetf for this situation review! 
I also saw Debian example but they have a note that `DEST_MODULE_LOCATION` is not important in their case.

OK, let's keep `/extra` until someone will provide better option working for majority of distros.

PS. I added information about all this and `depmod.d` in the README.md documentation.
The issue can be considered solved and close but if someone has anything to add - feel free to do so here.

closed 2020-08-30 by free5lot