free5lot/hid-apple-patched #68: iso_layout not working when the keyboard is connected via Bluetooth

daquinons, 2020-06-01:
I've successfully installed the driver using DKMS, but the parameter iso_layout doesn't take any effect. 
For example, ejectcd_as_delete works (not in all the input fields and applications, but at least the standard eject is not appearing every time that I missclick).

Anyway, when I connect the keyboard via usb, iso_layout is used. I can see it because the mapping of the >< key is correct.

Please, let me know if I can provide any debugging information.

I just got the Keyboard and I haven't updated the firmware because I don't have a Mac right now.

daquinons, 2020-06-01:
Another test that may give some information about what's going on.

- I connect the keyboard via USB, everything works as expected, including the correct mapping
- I run `sudo modprobe -r hid_apple`
- No key works
- `sudo modprobe hid_apple` and the key works again

- I connect the keyboard via Bluetooth, the keys are wrongly mapped
- I run `sudo modprobe -r hid_apple`
- The keys keep working, with the wrong mapping

ozdemir1419, 2020-06-07:
@daquinons Same Bug
This module is works great except two keys. [ The key below ESC ] and  [ The key next to the Left Shift ] mixed.
When I connect the keyboard via USB, everything works as expected.
When I connect the keyboard via Bluetooth, the keys are wrongly mapped.

I bought [ Apple Magic Keyboard 2, without numpad version, Turkish Q Layout ] two days ago.
I installed Arch Linux ( Turkish F Layout ) yesterday.

### <ins>LOGS</ins>
```
$ xev -event keyboard # WHEN I PRESS THE KEY BELOW ESC
KeyPress event, serial 28, synthetic NO, window 0x4c00001,
    root 0x5c2, subw 0x0, time 1471626, (553,469), root:(589,498),
    state 0x0, keycode 94 (keysym 0x3c, less), same_screen YES,
    XLookupString gives 1 bytes: (3c) "<"
    XmbLookupString gives 1 bytes: (3c) "<"
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x4c00001,
    root 0x5c2, subw 0x0, time 1471626, (553,469), root:(589,498),
    state 0x0, keycode 94 (keysym 0x3c, less), same_screen YES,
    XLookupString gives 1 bytes: (3c) "<"
    XFilterEvent returns: False

$ xev -event keyboard # WHEN I PRESS THE KEY NEXT TO THE LEFT SHIFT
KeyPress event, serial 25, synthetic NO, window 0x5000001,
    root 0x5c2, subw 0x0, time 1612400, (467,433), root:(503,462),
    state 0x0, keycode 49 (keysym 0x2b, plus), same_screen YES,
    XLookupString gives 1 bytes: (2b) "+"
    XmbLookupString gives 1 bytes: (2b) "+"
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x5000001,
    root 0x5c2, subw 0x0, time 1612460, (467,433), root:(503,462),
    state 0x0, keycode 49 (keysym 0x2b, plus), same_screen YES,
    XLookupString gives 1 bytes: (2b) "+"
    XFilterEvent returns: False

# dkms status
hid-apple, 1.0, 5.6.15-arch1-1, x86_64: installed
hid-apple-patched, 20200410.d5ffaff, 5.6.15-arch1-1, x86_64: installed (original_module exists)

# ls /sys/module/hid_apple/parameters/
ejectcd_as_delete  fnmode  iso_layout  rightalt_as_rightctrl  swap_fn_leftctrl	swap_opt_cmd

# cat /sys/module/hid_apple/parameters/iso_layout 
1

# uname -a
Linux arch 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000 x86_64 GNU/Linux

# cat /etc/modprobe.d/hid_apple.conf 
options hid_apple fnmode=2 swap_fn_leftctrl=1 swap_opt_cmd=1 rightalt_as_rightctrl=1 ejectcd_as_delete=1

# echo "0" > /sys/module/hid_apple/parameters/iso_layout # not working
# echo "1" > /sys/module/hid_apple/parameters/iso_layout # not working
# echo "0" > /sys/module/hid_apple/parameters/iso_layout # not working
# echo "1" > /sys/module/hid_apple/parameters/iso_layout # not working
```

## I applied these solutions, apple keyboard works without problems but this time laptop keys are mixing (I use apple keyboard with laptop)

In Arch Linux, apple:badmap does not work. Because some settings missing. I copied these files from Debian 10 Machine to Arch Linux Machine and replaced same destination.

```
/usr/share/X11/xkb/keycodes/macintosh
/usr/share/X11/xkb/rules/evdev
/usr/share/X11/xkb/rules/base
```

### <ins> Fix 1 Temporary </ins>
`setxkbmap -option apple:badmap`

### <ins> Fix 2 Permanent</ins>
```
$ cat /usr/share/X11/xorg.conf.d/00-keyboard.conf
Section "InputClass"
    Identifier "system-keyboard"
    MatchIsKeyboard "on"
    Option "XkbLayout" "tr"
    Option "XkbModel" "pc101"
    Option "XkbVariant" "f"
    Option "XkbOptions" "apple:badmap"
EndSection
```
### <ins>Here is how I installed this module</ins>

\# 1 
`pacman -S base-devel dkms linux-headers`

\# 2 This Aur package is dependency.
<a href="https://aur.archlinux.org/packages/mkinitcpio-modconf-hook-git/">aur.archlinux.org/packages/mkinitcpio-modconf-hook-git/</a>

\# 3 I think Aur package is recommended way 
<a href="https://aur.archlinux.org/packages/hid-apple-patched-git-dkms/">aur.archlinux.org/packages/hid-apple-patched-git-dkms/</a>

<hr>

Note: I installed Aur packages with `makepkg -si` command. And I did not run `mkinitcpio -p linux` command. Aur package automaticly updates initramfs.

free5lot, 2020-06-08:
`iso_layout` is a parameter from the original mainstream hid-apple and is not expected to act differently.
To be clear, how does `iso_layout` work with the original hid-apple (without this repo) with Bluetooth keyboard? Is it acting differently with mainstream kernel? Because according to the code it should not.

ozdemir1419, 2020-06-08:
@free5lot Same result. `iso_layout` not working with the original `hid-apple` (without this repo) with Bluetooth keyboard. [ Tested now with fresh Arch Linux installation ] It is not acting differently with mainstream kernel.

free5lot, 2020-06-08:
@ozdemir1419 that's what I've expected. The patched version should not affect `iso_layout` in any way.
I'm not sure this issue can be fixed in this repo, you probably should file the report to the mainstream bug-trackers or connect to the maintainers directly.
If it's fixed upstream it will be migrated here too as this repo is not modifying how `iso_layout` acts.

I searched a bit and it seems this problem exists in mainstream for a while:
[Params to hid_apple kernel module not recognized for Bluetooth Apple keyboard - Linux Mint Forums](https://forums.linuxmint.com/viewtopic.php?t=286855)

I will keep this issue open for a while so people be able to discuss it and maybe provide any ideas or links for bug reports.

daquinons, 2020-06-08:
I reported it here as well https://bugzilla.kernel.org/show_bug.cgi?id=99881

free5lot, 2020-06-08:
@daquinons thanks. Maybe they will fix.

**But we can try to do it too. One thing we can check easily is to modify one line, install via DKMS and check. Anyone with Bluetooth keyboard can do it?**

My idea would be modify this line:
https://github.com/free5lot/hid-apple-patched/blob/master/hid-apple.c#L289
by replacing
`  if (hid->country == HID_COUNTRY_INTERNATIONAL_ISO) {`
with
`  if (true) {`

Then test if `iso_layout` is working properly after it.
I have no such Bluetooth keyboard. Anyone?

Maybe for Bluetooth keyboards in Bluetooth mode for some reason `hid->country` is not equal to `HID_COUNTRY_INTERNATIONAL_ISO`.
This condition [was added recently (2017), previously it was different, take a look](https://patchwork.kernel.org/patch/9762845/).

ozdemir1419, 2020-06-08:
@free5lot THANKS! Now `iso_layout` working without problems with apple magic bluetooth keyboard without numpad version in Arch Linux.

free5lot, 2020-06-08:
@ozdemir1419 great, you are welcome!

We need the second confirmation and I will write to the upstream maintainer, actually it seems this problem was in the upstream kernel for about 3 years.

daquinons, 2020-06-08:
I'm running into an unrelated issue this time that I'm trying to build the module (after last tried I uninstalled it using dkms remove).

When I try to build it
```
[david@x1 hid-apple-patched]$ sudo dkms build hid-apple/1.0 --kernelsourcedir /usr/src/kernels/5.6.16-300.fc32.x86_64

Preparing kernel 5.6.15-300.fc32.x86_64 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make mrproper...(bad exit status: 2)
Warning: using /usr/src/kernels/5.6.16-300.fc32.x86_64/.config
(I hope this is the correct config for this kernel)
make oldconfig...(bad exit status: 2)
make prepare-all...(bad exit status: 2)

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.6.15-300.fc32.x86_64 -C /usr/src/kernels/5.6.16-300.fc32.x86_64 M=/var/lib/dkms/hid-apple/1.0/build...
cleaning build area...
cleaning kernel tree (make mrproper)...(bad exit status: 2)

DKMS: build completed.
```
I really can't remember if last time I had those bad exit status. It installs, aparently, successfully
```
[david@x1 hid-apple-patched]$ sudo dkms install hid-apple/1.0

hid-apple.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.6.15-300.fc32.x86_64/extra/hid-apple.ko.xz
   - Storing in /var/lib/dkms/hid-apple/original_module/5.6.15-300.fc32.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.6.15-300.fc32.x86_64/extra/
Adding any weak-modules

depmod....

Backing up initramfs-5.6.15-300.fc32.x86_64.img to /boot/initramfs-5.6.15-300.fc32.x86_64.img.old-dkms
Making new initramfs-5.6.15-300.fc32.x86_64.img
(If next boot fails, revert to initramfs-5.6.15-300.fc32.x86_64.img.old-dkms image)
dracut.......

DKMS: install completed.
```
But modprobe gives this error
```
[david@x1 hid-apple-patched]$ sudo modprobe -r hid_apple; sudo modprobe hid_apple
modprobe: ERROR: could not insert 'hid_apple': Exec format error
```

ozdemir1419, 2020-06-08:
@daquinons I succeed by uninstalling and reinstalling dkms module. I did these things:

Unload Module
`# modprobe -r hid_apple`

List all dkms modules with
`# dkms status`

Then, remove all dkms modules
`# dkms remove hid-apple/1.0 --all`
`# dkms remove hid-apple-patched/20200608.3afbab7 --all` this code can be different in your case (3afbab7)

Then, list remaining modules and continue uninstalling modules
`# dkms status`

I think you should clean old module source code like I did.
```
# ls /var/lib/dkms/ # check any hid-apple folder and delete all hid-apple folders 
# ls /usr/src/ # continue deleting hid-apple folders
```

Reboot and check mainstream kernel module loaded 
` # modinfo hid_apple` It should show 3 `parm` values. If it shows 6 `parm` values, we failed uninstalling patched module. Do not continue. Go to step 1 or get help.

Edit `hid-apple.c` file. Just replace line 289 with `if (true) {`

### <ins>Example</ins>
```c
...
287
288    if (iso_layout) {
289        if (true) {
290            trans = apple_find_translation(apple_iso_keyboard, usage->code);
291            if (trans) {
292                    input_event(input, usage->type, trans->to, value);
293                    return 1;
294            }
295        }
296    }
297
...
```

Load fixed module
`modprobe -r hid_apple ; modprobe hid_apple`

Check `iso_layout` working

`echo "1" > /sys/module/hid_apple/parameters/iso_layout`

If it is not working I think you should reinstall your linux distribution.

daquinons, 2020-06-08:
> @daquinons I succeed by uninstalling and reinstalling dkms module. I did these things:
> 
> Unload Module
> `# modprobe -r hid_apple`
> 
> List all dkms modules with
> `# dkms status`
> 
> Then, remove all dkms modules
> `# dkms remove hid-apple/1.0 --all`
> `# dkms remove hid-apple-patched/20200608.3afbab7 --all` this code can be different in your case ([3afbab7](https://github.com/free5lot/hid-apple-patched/commit/3afbab7ec02d7ffbd119dca15dcddd4239649a7a))
> 
> Then, list remaining modules and continue uninstalling modules
> `# dkms status`
> 
> I think you should clean old module source code like I did.
> 
> ```
> # ls /var/lib/dkms/ # check any hid-apple folder and delete all hid-apple folders 
> # ls /usr/src/ # continue deleting hid-apple folders
> ```
> 
> Reboot and check mainstream kernel module loaded
> ` # modinfo hid_apple`
> It should show 3 `parm` values. If it shows 6 `parm` values, we failed uninstalling patched module. Do not continue. Go to step 1 or get help.
> 
> Edit `hid-apple.c` file. Just replace line 289 with `if (true) {`
> ### Example
> 
> ```c
> ...
> 287
> 288    if (iso_layout) {
> 289        if (true) {
> 290            trans = apple_find_translation(apple_iso_keyboard, usage->code);
> 291            if (trans) {
> 292                    input_event(input, usage->type, trans->to, value);
> 293                    return 1;
> 294            }
> 295        }
> 296    }
> 297
> ...
> ```
> 
> Install fixed module
> 
> ```shell
> # dkms add .
> # dkms build hid-apple/1.0
> # dkms install hid-apple/1.0
> ```
> 
> Load fixed module
> `# modprobe -r hid_apple ; modprobe hid_apple`
> 
> Check `iso_layout` working
> 
> `# echo "1" > /sys/module/hid_apple/parameters/iso_layout`
> 
> If it is not working I think you should reinstall your linux distribution.

It works now! <> :) Thank you both for your help @free5lot @ozdemir1419, I think this is ready for an upstream fix!

free5lot, 2020-06-14:
@daquinons so, You will take this Bluetooth-related issue from here? Your message with this offer disappeared for some reason.

daquinons, 2020-06-15:
Yep, but I think is going to take a bit longer because I reported the findings to Fedora kernel contributors (as I'm definitely not capable to make a good-enough PR myself).

free5lot, 2020-06-24:
I send a report about this issue to the up-stream kernel maintainer **Jiri Kosina**. I put details about possible fix and etc.

free5lot, 2020-08-21:
I'm going to close this issue as it's upstream-code related and not about patched version's code.
I sent details to `Jiri Kosina <jkosina@suse.cz>` and consider my part to be done on this matter.
I hope @daquinons will follow and pursuit the actual issue till it's fixed.

closed 2020-08-21 by free5lot