free5lot/hid-apple-patched #71: Keychron connected via bluetooth not working

paulogr, 2020-08-27:
Hi, when trying to use my k2 via bluetooth my Pop-os just use hid-generic instead of the hid-apple version I installed.

It's installed

paulogr@desk:~$ sudo dkms status
[sudo] password for paulogr: 
hid-apple, 1.0, 5.4.0-7634-generic, x86_64: installed

But my OS set hid-generic as driver

paulogr@desk:~$ dmesg | grep -i bluetooth
[    8.971118] hid-generic 0005:05AC:024F.0005: input,hidraw4: BLUETOOTH HID v1.1b Keyboard [Keychron K2] on 0c:7a:15:fa:47:08

Could someone help me?

Thanks

free5lot, 2020-08-28:
What happens if you use original hid-apple module, is it used instead of hid-generic?

paulogr, 2020-08-28:
How can I switch back to original hid-apple?

Just need to remove the patched version?

free5lot, 2020-08-28:
Yes, probably something like:
`sudo dkms remove hid-apple/1.0`

hyperfocus1337, 2020-08-29:
Coincidentally I have the same keyboard, when running `sudo dmesg | grep -i bluetooth` this is the output:

```
[   12.977011] Bluetooth: Core ver 2.22
[   12.977029] Bluetooth: HCI device and connection manager initialized
[   12.977032] Bluetooth: HCI socket layer initialized
[   12.977034] Bluetooth: L2CAP socket layer initialized
[   12.977037] Bluetooth: SCO socket layer initialized
[   16.377060] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.377062] Bluetooth: BNEP filters: protocol multicast
[   16.377066] Bluetooth: BNEP socket layer initialized
[   22.961108] Bluetooth: RFCOMM TTY layer initialized
[   22.961117] Bluetooth: RFCOMM socket layer initialized
[   22.961121] Bluetooth: RFCOMM ver 1.11
[  753.855852] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  753.855859] Bluetooth: HIDP socket layer initialized
[  754.875946] input: Keychron K2 Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:69/0005:05AC:024F.0009/input/input27
[  754.876288] input: Keychron K2 Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:69/0005:05AC:024F.0009/input/input28
[  754.876414] input: Keychron K2 System Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:69/0005:05AC:024F.0009/input/input29
[  754.876579] hid-generic 0005:05AC:024F.0009: input,hidraw6: BLUETOOTH HID v1.1b Keyboard [Keychron K2] on 00:1a:7d:da:71:11
```

It does work for me when I have the keyboard connected to bluetooth, however mysteriously enough I can't get it to work in cable mode.

I'm using Manjaro linux in combination with the Windows mode on the keyboard (opposed to the Mac mode). This is the kernel I'm using when running `dkms status`: hid-`apple-patched, 20200610.78148d5, 5.8.1-3-MANJARO, x86_64: installed (original_module exists)`

@free5lot thanks for excellent work on this patch, where can I donate?

free5lot, 2020-08-29:
@hyperfocus1337 and @paulogr you may try the version updated for the 5.8 kernel: take a loot at pull request #70. 
If it works for you I will accept PR #70 as it seems to be fine, the only block there is a need of second opinion - I'm waiting for someone with external keyboard to check if PR #70 breaks nothing for them.

@hyperfocus1337 thank you but currently no donations are possible.
The whole project is just a small set of patches/features and half of those were done by community via PRs.

hyperfocus1337, 2020-08-29:
@free5lot the MR seems interesting, I would like to test it but I don't know how to easily install it, tried adjusting the AUR pkgbuild file but I can't use it with a forked repository. Any suggestions on how to easily install it on Manjaro/AUR?

```
==> ERROR: /var/tmp/pamac-build-user/hid-apple-patched-git-dkms/hid-apple-patched is not a clone of https://github.com/jbcjorge/hid-apple-patched
    Aborting...
Failed to build hid-apple-patched-git-dkms
```

free5lot, 2020-08-29:
@hyperfocus1337 
Well, the PR #70 seems to be fine as I reviewed it and found nothing bad. I simply can check it with 5.8 myself so I hold it.
I hope @Aetf himself as a AUR package creator or @almson as another savvy Arch user can help you.

Aetf, 2020-08-29:
@hyperfocus1337 that error is caused by your aur helper (pamac I assume from the path) still having the old repo somewhere in the cache. Delete that and retry. I never used pamac but my guess is it's somewhere in $HOME/.cache

hyperfocus1337, 2020-08-30:
After uninstalling the hid-apple-patched-git-dkms package and rebooting my machine I noticed I could still use my Keychron K2 over bluetooth with a working left ctrl key (instead of it remapping by default to a fn key).

In my situation, in order to properly use the Keychron K2 with the keys in the expected location on Linux, I need to use the Windows mode with a bluetooth connection. It appears I do not have to install the patch in this repo to make this work.

This is the key layout I have on my Keychron K2 (default Mac key layout).
[image]

These are the default (and desired) keymappings of the Keychron K2 on my Manjaro linux machine (using Windows mode over a bluetooth connection maps to my desired key locations).
[image]

However, I was hoping I could properly use the keyboard in cable mode so I don't have to connect the keyboard with bluetooth all the time (as Linux doesn't automatically reconnect the keyboard, like Mac does). I was hoping this patch could solve that issue for me.

It appears the patch by default doesn't solve this issue for me, neither does the MR from @jbcjorge, so I cannot tell whether this PR causes something to break as the patch doesn't work for me (personally) to begin with.

This is how I installed PR https://github.com/free5lot/hid-apple-patched/pull/70. These are the steps I took
1. Downloading PKGBUILD: `yay -G hid-apple-patched-git-dkms`
2. Editing PKGBUILD to include PR repo and branch (https://github.com/jbcjorge/hid-apple-patched/tree/linux_5.8.x)
3. Install the package: `makepkg -si`
4. Rebooted my machine
5. The correct repo is shown in pamac (see screenshot)
6. Left ctrl is still fn key in cabled mode (using Windows mode on Keychron K2)

[image]

Running `sudo dkms status` shows me the expected output:
````
hid-apple-patched, 20200820.f5b0961, 4.14.194-2-MANJARO, x86_64: installed (original_module exists)
hid-apple-patched, 20200820.f5b0961, 4.19.141-2-MANJARO, x86_64: installed (original_module exists)
hid-apple-patched, 20200820.f5b0961, 5.4.60-2-MANJARO, x86_64: installed (original_module exists)
hid-apple-patched, 20200820.f5b0961, 5.7.17-2-MANJARO, x86_64: installed (original_module exists)
hid-apple-patched, 20200820.f5b0961, 5.8.3-2-MANJARO, x86_64: installed (original_module exists)
vboxhost, 6.1.12_non_OSE, 4.14.194-2-MANJARO, x86_64: installed
vboxhost, 6.1.12_non_OSE, 4.19.141-2-MANJARO, x86_64: installed
vboxhost, 6.1.12_non_OSE, 5.4.60-2-MANJARO, x86_64: installed
vboxhost, 6.1.12_non_OSE, 5.7.17-2-MANJARO, x86_64: installed
vmware-workstation, 15.5.6_16341506, 4.14.194-2-MANJARO, x86_64: installed
vmware-workstation, 15.5.6_16341506, 4.19.141-2-MANJARO, x86_64: installed
vmware-workstation, 15.5.6_16341506, 5.4.60-2-MANJARO, x86_64: installed
vmware-workstation, 15.5.6_16341506, 5.7.17-2-MANJARO, x86_64: installed
````

When running `ls /sys/module/hid_apple/parameters/` the options are shown:
```
ejectcd_as_delete  fnmode  iso_layout  rightalt_as_rightctrl  swap_fn_f13_insert  swap_fn_leftctrl  swap_opt_cmd
```

Output of `cat /sys/module/hid_apple/parameters/swap_fn_leftctrl` is `1`.

Unsure how to move forward in making this work.

free5lot, 2020-08-30:
> After uninstalling the `hid-apple-patched-git-dkms` package and rebooting my machine I noticed I could still use my Keychron K2 over bluetooth with a working left ctrl key (instead of it remapping to a fn key by default). So It appears I do not have to install the patch in this repo in order to properly use my keyboard.
Yes, starting from Kernel 5.8 it already has `swap_fn_leftctrl` parameter from our patched version.
I submitted this flag to the maintainer in May 2020 (read header of the README) and it was expected to get to 5.8 kernel. :)
Other options like `swap_opt_cmd` and `rightalt_as_rightctrl` are still only in patched version. But pushing `swap_fn_leftctrl`  upstream went well, so, maybe all those will be accepted too as I send them later.

About the rest, I'm a bit confused. This keyboard of yours Keychron K2 supports 2 modes and switch on the hardware level? Or how exactly do you switch between Mac-Windows modes in GNU/Linux?

hyperfocus1337, 2020-08-30:
Thanks for getting this integrated upstream, so I can now enjoy my keyboard without struggling.

> About the rest, I'm a bit confused. This keyboard of yours Keychron K2 supports 2 modes and switch on the hardware level? Or how exactly do you switch between Mac-Windows modes in GNU/Linux?

Yes, the keyboard has two hardware switches, one to switch from bluetooth to cable mode and another switch to switch from Windows/Mac mode.
[image]

free5lot, 2020-08-30:
I see. Than it can be a completely different story.
I'm not sure if the current Kernel support this keyboard always with hid-apple in all modes and connection types and if it deals with fn key properly. Patch version relies on all this from the original code and only add minor useful tweaks possible not modifying the core code too much.

hyperfocus1337, 2020-08-30:
> I'm not sure if the current Kernel support this keyboard always with hid-apple in all modes and connection types and if it deals with fn key properly.
It's ok, I can use it in bluetooth mode for now, and I wouldn't eventually need it to work with all different combinations. I created the spreadsheet to figure out which keys were mapped by default to begin with, so I could figure out a strategy to decide which keys to remap in order to reach the desired result. Thanks for your help regardless.

vladvasiliu, 2020-09-15:
Hi @hyperfocus1337, I have the same keyboard and almost same distro (Arch) but in my case the wired mode works as expected (I actually don't have BT to test that mode).

I'm a bit confused by this line in your post above: 

> Output of `cat /sys/module/hid_apple/parameters/swap_fn_leftctrl` is `1`.

For me that means "left ctrl is fn; fn is left ctrl". Have you tried setting that to 0?

Using the keyboard in Mac mode and the standard hid_apple module (as of linux 5.8) and the settings below I get your desired layout:

```
fnmode:           2
iso_layout:       1
swap_fn_leftctrl: 0
swap_opt_cmd:     1
```

hyperfocus1337, 2020-09-15:
Whoa, you are right, changing `/sys/module/hid_apple/parameters/swap_fn_leftctrl` to `0` does solve my challenge!

Thanks for pointing that out, I'm using the keyboard in cable mode now (with Mac mode).

Awesome!

closewait, 2021-08-19:
@paulogr pr #74  should fix the module issue. My k8 was behaving like your K2 (ie load hid-apple in wire mode but hid-generic in bluetooth).

closewait, 2021-08-19:
Hi @paulogr 

To load hid-apple instead of hid-generic in bluetooth mode for your k2, you can take the change from this [commit](https://github.com/closewait/hid-apple-patched/commit/ff77d9b18174bc9a04dc8f8d5479b04e592ec77d#diff-ad0d321a3233b309197a30bb836af2fde04765d0813879c380e7b43f7e05f44b).

I had the same issue with my k8.

almson, 2021-08-20:
> Hi @paulogr
>
> To load hid-apple instead of hid-generic in bluetooth mode for your k2, you can take the change from this [...]

That patch will soon be in Linux 5.14, and I've created a PR for this repo https://github.com/free5lot/hid-apple-patched/pull/76