free5lot/hid-apple-patched #79: Help Magic Keyboard Touch ID and Ubuntu 21.10

cue108, 2021-11-19:
I simply don't know how to configure it.
The module is installed via the first instructions in the Readme.md and I have reboot.

The module is not even loaded:

`lsmod | grep hid
hid_magicmouse         20480  0
hidp                   32768  2
bluetooth             663552  42 btrtl,hidp,btintel,btbcm,bnep,btusb,rfcomm
mac_hid                16384  0
hid_generic            16384  0
usbhid                 61440  0
hid                   135168  4 hidp,usbhid,hid_generic,hid_magicmouse
`

If I do a modprobe it looks fine but I think the module is not used as I cannot identify any change in the keyboard usage:

`sudo modprobe -v hid_apple
insmod /lib/modules/5.13.0-21-generic/updates/dkms/hid-apple.ko fnmode=2 swap_opt_cmd=1 rightalt_as_rightctrl=1 ejectcd_as_delete=1 swap_fn_f13_insert=1 
`

I have a Magic Keyboard with Touch ID together with a magic mouse and I am running Ubuntu 21.10 on a PC.

Any hint is appreciated!

d0nth4x, 2021-11-19:
Look like same problem as mine. You have to check your product id and add it like me. Check PR #77 .
idk what is prod id for keyboard with touch id so you have to use tool like Hardinfo and check it.
Hope it helps ;)

cue108, 2021-11-20:
Thanx a lot. I pulled your fork and it didn't work but I could find the USB id by checking syslog.

Maybe you want to add your PR with the following address in hid-ids.h:

`
#define USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_touch_id_2021 0x029f
`

and add the following lines to hid-apple.c:

```
	{ HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_touch_id_2021),
		.driver_data = APPLE_HAS_FN },
	{ HID_BLUETOOTH_DEVICE(BT_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_touch_id_2021),
		.driver_data = APPLE_HAS_FN },
```

It works great!

sergei-sss, 2021-11-25:
@cue108 @d0nth4x 

Guys, how is it possible to find device id like "0x02.." ?

I tried using two methods but got nothing similar:

1) lsusb
`Bus 003 Device 008: ID 05ac:029c Apple, Inc. Magic Keyboard`

2) tail -f /var/log/syslog and dettach/attach keyboard

`Nov 25 13:40:45 omen kernel: [ 3053.153856] usb 3-1: USB disconnect, device number 8
Nov 25 13:40:45 omen acpid: input device has been disconnected, fd 19
Nov 25 13:40:45 omen /usr/libexec/gdm-x-session[2917]: (II) config/udev: removing device Apple Inc. Magic Keyboard
Nov 25 13:40:45 omen /usr/libexec/gdm-x-session[2917]: (**) Option "fd" "84"
Nov 25 13:40:45 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: device removed
Nov 25 13:40:45 omen /usr/libexec/gdm-x-session[2917]: (II) UnloadModule: "libinput"
Nov 25 13:40:45 omen /usr/libexec/gdm-x-session[2917]: (II) systemd-logind: releasing fd for 13:72
Nov 25 13:40:55 omen wpa_supplicant[1193]: wlo1: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-63 noise=9999 txrate=585100
Nov 25 13:40:59 omen kernel: [ 3067.137851] usb 3-1: new full-speed USB device number 11 using xhci_hcd
Nov 25 13:40:59 omen kernel: [ 3067.306300] usb 3-1: New USB device found, idVendor=05ac, idProduct=029c, bcdDevice= 3.20
Nov 25 13:40:59 omen kernel: [ 3067.306310] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 25 13:40:59 omen kernel: [ 3067.306313] usb 3-1: Product: Magic Keyboard
Nov 25 13:40:59 omen kernel: [ 3067.306316] usb 3-1: Manufacturer: Apple Inc.
Nov 25 13:40:59 omen kernel: [ 3067.306318] usb 3-1: SerialNumber: F0T1307002S18DTAA
Nov 25 13:40:59 omen kernel: [ 3067.346611] hid-generic 0003:05AC:029C.000F: hiddev1,hidraw4: USB HID v1.10 Device [Apple Inc. Magic Keyboard] on usb-0000:06:00.3-1/input0
Nov 25 13:40:59 omen kernel: [ 3067.357775] input: Apple Inc. Magic Keyboard as /devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb3/3-1/3-1:1.1/0003:05AC:029C.0010/input/input42
Nov 25 13:40:59 omen kernel: [ 3067.417861] hid-generic 0003:05AC:029C.0010: input,hiddev2,hidraw5: USB HID v1.10 Keyboard [Apple Inc. Magic Keyboard] on usb-0000:06:00.3-1/input1
Nov 25 13:40:59 omen mtp-probe: checking bus 3, device 11: "/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb3/3-1"
Nov 25 13:40:59 omen mtp-probe: bus: 3, device: 11 was not an MTP device
Nov 25 13:40:59 omen upowerd[1545]: treating change event as add on /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb3/3-1
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) config/udev: Adding input device Apple Inc. Magic Keyboard (/dev/input/event8)
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Apple Inc. Magic Keyboard: Applying InputClass "libinput keyboard catchall"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) Using input driver 'libinput' for 'Apple Inc. Magic Keyboard'
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) systemd-logind: got fd for /dev/input/event8 13:72 fd 97 paused 0
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Apple Inc. Magic Keyboard: always reports core events
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "Device" "/dev/input/event8"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "_source" "server/udev"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: is tagged by udev as: Keyboard
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: device is a keyboard
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: device removed
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb3/3-1/3-1:1.1/0003:05AC:029C.0010/input/input42/event8"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) XINPUT: Adding extended input device "Apple Inc. Magic Keyboard" (type: KEYBOARD, id 9)
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "xkb_model" "pc105"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "xkb_layout" "pl,ru"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "xkb_variant" ","
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (**) Option "xkb_options" "grp_led:scroll"
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: is tagged by udev as: Keyboard
Nov 25 13:40:59 omen /usr/libexec/gdm-x-session[2917]: (II) event8  - Apple Inc. Magic Keyboard: device is a keyboard
Nov 25 13:40:59 omen mtp-probe: checking bus 3, device 11: "/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb3/3-1"
Nov 25 13:40:59 omen mtp-probe: bus: 3, device: 11 was not an MTP device`

cue108, 2021-11-25:
It's called the product ID. In your syslog it even explicitly says it
"idProduct=029c" but you can also use the usb device id, just use the
last 4 digits after the ':'.
alos your hidID works: hid-generic 0003:05AC:029C.0010

It's alway 029C in you case.
cheers

siema001, 2022-02-13:
In my case it's 029a

joshuayanggithub, 2025-06-18:
Did you guys get touch ID to work with Ubuntu in the end?