free5lot/hid-apple-patched #80: linnux 5.15.5-1 on debian with dkms: iso_layout stopped working

kapouer, 2021-12-03:
Hi,
i usually set iso_layout to 1, and it sticks:
```
root@lal:~# cat /sys/module/hid_apple/parameters/iso_layout
1
```
but after rebooting to latest kernel the "@" and "<" keys are wrong.
I checked hid-apple is correctly loaded:
```
dkms status
hid-apple/1.0, 5.14.0-4-amd64, x86_64: installed
hid-apple/1.0, 5.15.0-2-amd64, x86_64: installed
```

and dmesg doesn't show anything interesting.
(also, I've tried dkms unbuild, remove, and add build again, no luck).

free5lot, 2022-01-13:
Hi. Do you have `/sys/module/hid_apple/parameters/iso_layout` changed after reboot? Do you use config file for `hid-apple` to set this value?

kapouer, 2022-01-13:
> cat /sys/module/hid_apple/parameters/iso_layout
> 1

no i use `/etc/sysfs.d/10-hidapple.conf` with
```
module/hid_apple/parameters/iso_layout = 1
module/hid_apple/parameters/fnmode = 1
```

I worked around this by exchanging the two keys in xkb's ~/.config/xkb/

free5lot, 2022-01-14:
And the original (not patched) `hid-apple` module has different behavour with `iso_layout`? Are you sure?
Options `fnmode` and `iso_layout` are part of the original hid-apple module.

kapouer, 2022-01-14:
No ! Same thing with the original hid-apple.

free5lot, 2022-01-14:
> No ! Same thing with the original hid-apple.
Well, in that case the problem is not related to this repo (patched version). I recommend you to report your proplem with all details to upstream hid-apple maintainers as they know better what can be wrong with `iso_layout`.

I'm closing this issue, but feel free to comment here or ask to reopen if necessary.

free5lot, 2022-01-14:
P.S. I fixed quote in my last message, please re-read.

closed 2022-01-14 by free5lot

kapouer, 2022-02-10:
FYI it's been fixed in linux 5.16.7

free5lot, 2022-04-13:
> FYI it's been fixed in linux 5.16.7

Thanks, good to know.