free5lot/hid-apple-patched #86: M1 Macbook air support planned?

gronka, 2022-08-13:
I tried this with M1 macbook air and it did not work. I'd love to have it. I'm interested in learning how to write kernel modules if you have leads on where to start, but I'm not sure when I'd get to it.

free5lot, 2022-10-03:
I'm sorry, I have no M1 device and no expertise in that architecture. Also I don't know how Linux kernel is going to support it in general (different source tree, git branch, conditional-builds with macro or something else?). Do you have any information?

gronka, 2022-10-31:
All I can say is the Asahi linux project has come a long way and is working well. I use it for work; I read that Torvalds has also. So, I would try and apply the patch to that.

captn3m0, 2023-01-18:
Using a M2 with Asahi, and most of the requirements are already fulfilled by the hid_apple shipped with Asahi. The only exception is `rightalt_as_rightctrl`, which is still missing from the upstream hid_apple module.

free5lot, 2023-01-19:
@gronka @captn3m0 related question: https://github.com/free5lot/hid-apple-patched/issues/1#issuecomment-1396466725

fvsnippets, 2023-10-08:
> Using a M2 with Asahi, and most of the requirements are already fulfilled by the hid_apple shipped with Asahi. The only exception is `rightalt_as_rightctrl`, which is still missing from the upstream hid_apple module.

Haven't tried using M1/M2 chip (I do not own the hardware), but maybe #95 could be related ? Also, there I've posted a possible workaround.