free5lot/hid-apple-patched #87: Keyboard Disabled Fedora 34

Lucrecious, 2022-12-04:
After I ran this comment from the github page:
`sudo modprobe -r hid_apple; sudo modprobe hid_apple`

I thought running this with the `;` would reenable the keyboard.

My keyboard was disabled, how do I turn it back on?

baack, 2022-12-04:
@Lucrecious you can simply reboot you PC and it will use the default hid_apple module if you were unloading the original one manually with `modprobe`.

Lucrecious, 2022-12-04:
The first thing I tried was rebooting the PC but nothing changed. The keyboard is still unresponsive - very weird!

Luckily, this is a fresh Fedora install, so worst case scenario I can just re-install.

free5lot, 2022-12-04:
You need to provide more information. What you did, what hardware/macbook do you have and etc.

Based on you messages I suppose you have installed the patched version via DKMS before running this command., haven't you? If yes, then the modprobe command is not the reason of problems, but the fact that patched version of hid-apple module does not work properly in your case for some reason.

Before reinstall or something - try to connect USB keyboard and check if it works. This way you can remove installed DKMS module and get the default one working again in the worst case scenario.

Lucrecious, 2022-12-05:
I'm on OSX
CPU: 2 GHz Quad-Core Intel Core i5
RAM: 16GB
Graphics Card: Intel Iris Plus Graphics 1536 MB

I went through the install via DKMS instructions on the github page, up until this step:
`sudo update-initramfs -u`

This command wasn't available on Fedora 37, and instead I used this instruction:
`dracut -f`

After that I tried to run:
`sudo modprobe -r hid_apple; sudo modprobe hid_apple`

And that's when my keyboard stopped working.

Also, unfortunately, I do not currently have an extra keyboard available to plug in. It's weird because even the virtual keyboard on the desktop doesn't work, but the virtual keyboard during login, does.