free5lot/hid-apple-patched #88: Apple Keyboard with Touch ID 2021 Product ID not included

Woz4tetra, 2022-12-17:
I wasn't able to get my keyboard to work until I added these lines:

`hid-apple.c`

```diff
                .driver_data = APPLE_HAS_FN },
        { HID_BLUETOOTH_DEVICE(BT_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_2021),
                .driver_data = APPLE_HAS_FN },
+       { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_TOUCH_ID_2021),
+               .driver_data = APPLE_HAS_FN },
+       { HID_BLUETOOTH_DEVICE(BT_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_TOUCH_ID_2021),
+               .driver_data = APPLE_HAS_FN },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_ANSI),
                .driver_data = APPLE_HAS_FN },
        { HID_BLUETOOTH_DEVICE(BT_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_ANSI),
```
`hid-ids.h`
```diff
 #define USB_DEVICE_ID_APPLE_MAGICTRACKPAD      0x030e
 #define USB_DEVICE_ID_APPLE_MAGICTRACKPAD2     0x0265
 #define USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_2021 0x029c
+#define USB_DEVICE_ID_APPLE_MAGIC_KEYBOARD_TOUCH_ID_2021 0x029f
 #define USB_DEVICE_ID_APPLE_FOUNTAIN_ANSI      0x020e
 #define USB_DEVICE_ID_APPLE_FOUNTAIN_ISO       0x020f
 #define USB_DEVICE_ID_APPLE_GEYSER_ANSI        0x0214
```

I noticed that the device was being loaded with `hid-generic` in `dmesg` logs until I rebuilt the module with these lines added.

Also, not sure if this is the correct way to do it but in order to get the module to load at boot, I added `hid_apple` the `/etc/modules` file. Otherwise, I'd have to run `sudo modprobe -r hid_apple; sudo modprobe hid_apple` after logging in.

This is the exact product in question:
https://www.apple.com/shop/product/MMMR3LL/A/magic-keyboard-with-touch-id-and-numeric-keypad-for-mac-models-with-apple-silicon-us-english-black-keys

Woz4tetra, 2022-12-17:
Related issue: https://github.com/free5lot/hid-apple-patched/issues/79

free5lot, 2023-01-19:
Thank you for feedback, I will sync the patched project with upstream to support it.