free5lot/hid-apple-patched #89: doesnt work on Apple MacBook Pro 13" - A1278 - Mid 2012

BlackMage2, 2023-02-26:
the function keys doesnt work on keyboard

BlackMage2, 2023-02-26:
some other keys don't work either

BlackMage2, 2023-02-26:
okay, it works now. 'options hid_apple fnmode=2' in /etc/modprobe.d/hid_apple.conf was the trick

BlackMage2, 2023-03-01:
but keyboard backlight control doesn't work

BlackMage2, 2023-03-01:
any idea how i can make that keyboard backlight works?

BlackMage2, 2023-03-01:
okay, i have now installed pommed and most keys work

BlackMage2, 2023-03-01:
but the eject key doesn't work, in xev when i press the eject key only 
`KeymapNotify event, serial 48, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0`
is written in xev

free5lot, 2023-04-22:
> any idea how can i enable the eject cd function?

I have no device with `Eject CD` button, so maybe somebody who uses this feature can help you somehow.

Keyboard backlight should not be affected by this patched version, it manages rebinds and swaps only.

BlackMage2, 2023-09-10:
correct, but when i install pommed, the keyboard backlight works. :D

BlackMage2, 2025-09-13:
> > any idea how can i enable the eject cd function?
> 
> I have no device with `Eject CD` button, so maybe somebody who uses this feature can help you somehow.
> 
> Keyboard backlight should not be affected by this patched version, it manages rebinds and swaps only.

oh, that is bad, on my device are only confusing key signals detected. eject media key doesnt recognized
my eject cd is right from the f12-key(=volume up function key). All functions works, only eject cd key is not recognized..