free5lot/hid-apple-patched #92: fnmode 2 doesn't work

SteppinKurry, 2023-04-20:
Hey,
I'm on a 2020 MacBook Pro with a Touchbar running Ubuntu 22.10. I have hid-apple-patched setup, and everything seems to work well except for the fnmode setting. My hid_apple.conf file looks exactly like the sample one in the readme, and all of the other changes work (swap fn and control, swap option and command, etc.), but no matter what I set it to, fnmode does not seem to make any changes. When it's all said and done, I'm hoping to have the function keys show by default, and only show the media keys when I press the fn key (technically the left control key). Any ideas?
Thanks in advance

free5lot, 2023-04-22:
Does the original `hid-apple` module works with `fnmode` in your case as expected?

SteppinKurry, 2023-04-23:
I tried hid-apple-3.19 by SicVolo, which gave me a lot of compilation errors. I tried HID-Apple from JanmanX, and it didn't seem to do anything for me. I enabled it with the given terminal commands, and it pretty much broke the keyboard (keyboard became entirely unresponsive, had to use external). So, no luck with either of those.

free5lot, 2023-04-23:
I mean Linux has its `hid-apple` out of the box (without installation of anything). The current version of it even has a feature accepted from my repo (to swap fn and ctrl), and it supports different `fnmode` too. Please try `fnmode` on the original version.

SteppinKurry, 2023-04-23:
To be perfectly honest, I still don't really understand what I'm doing. But, I found a script on T2 Linux (https://wiki.t2linux.org/guides/postinstall/) that fixed the problem for me, and now everything works exactly how I want it to. Thank you so much for your help with this!

closed 2023-04-23 by SteppinKurry