free5lot/hid-apple-patched #95: rightalt-as-rightctrl option not working

fvsnippets, 2023-10-08:
Hi!
I am pretty sure that *rightalt-as-rightctrl* option used to work with my old installation (was debian 11 "bullseye"/kernel 5.10 but I don't know the hid-apple-patched commit I was using then).

Now is broken with debian 12 "buster"/kernel 6.1 and latest hid-apple-patched commit on master [46f3457](https://github.com/free5lot/hid-apple-patched/tree/46f3457beb4909536fd7cceeb1b211d488aab5a5)

>    root@hostname# evtest
```
...snipped out...
Event: time 1696729013.627356, -------------- SYN_REPORT ------------
Event: time 1696729014.499395, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e6
Event: time 1696729014.499395, type 1 (EV_KEY), code 126 (KEY_RIGHTMETA), value 1
...snipped out...
```

My hardware: macbook air 2017 - spanish keyboard.
>    bustype: 0003
>    product: 0291
>    vendor: 05ac
>    version: 0110

Changing other options (following #19 issue) doesn't seems to work.
Other options (those I choose to use: *fnmode*, *swap_fn_leftctrl*, *swap_opt_cmd*) are working fine.

Thank you!

fvsnippets, 2023-10-08:
Notice: as a temporal workaround I am using udev.

>    root@hostname:\~# vi /etc/udev/hwdb.d/99-macbook-keyboard.hwdb
```
# Reassign RIGHT-ALT physical key as a temporal workaround to https://github.com/free5lot/hid-apple-patched/issues/95
evdev:input:b0003v05ACp0291e0110*
 KEYBOARD_KEY_700e6=rightctrl
```

>    root@hostname:\~# udevadm --debug hwdb --update
>    root@hostname:\~# udevadm \-\-debug trigger

For those interested in this solution, you can read this related [article](https://yulistic.gitlab.io/2017/12/linux-keymapping-with-udev-hwdb/) (but there's a lot information on google :P  )

abdoulayeYATERA, 2024-04-20:
> Notice: as a temporal workaround I am using udev.
>
>
> ```
> # Reassign RIGHT-ALT physical key as a temporal workaround to https://github.com/free5lot/hid-apple-patched/issues/95 [...]

Thanks

gpiat, 2025-10-23:
I have the same problem on my MacBook Pro 16" 2019 with T2-Linux Mint 22.2. This behavior is unaffected by the value of ` swap_opt_cmd`.